bambulab/BambuStudio #7992: PATest校准描述与实际功能不符

hack86, 2025-08-26:
### Bambu Studio Version

2.2.0.85

### Where is the application from?

Bambu Lab Official website

### OS version

Win11

### Printer

P1S

### How to reproduce

[image]

### Actual results

输入PA步长0.001时，提示有效值必须≥0.001，改成0.0015或0.002时正常了。

### Expected results

希望支持0.001的步长。提示也确实允许0.001，但实际，无法使用0.001。

### Project file & Debug log uploads

无需文件，打开最新版，准备页面，创建校准，选择压力提升，然后设置步长为0.001

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-BamBu, 2025-08-26:
@hack86 Thank you for your feedback. This issue has been resolved and verified in the internal version. Please look forward to the next version.

BoBoYu-BamBu, 2025-08-27:
@hack86 Hello, please update to our latest version V2.2.1.58 to verify this issue. The download link is: https://bambulab.cn/zh-cn/download/studio