bambulab/BambuStudio #7994: Missing setting

HegyLako-eredeti, 2025-08-26:
### Bambu Studio Version

newest

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambu lab A1 mini combo

### How to reproduce

1. Import a file with multiple models
2. Agree to import as multiple models
3. Try to use a slab for base under the other object
4. Try to find where to disable the "drop on the desk"-setting
5. Write a bug report

### Actual results

I cant find this setting in Move menu, in the menu from the file menu or at nowhere.

### Expected results

I want to be able to place objects (2mm over the desk) "in the air".

### Project file & Debug log uploads

[Teli platoniak.zip](https://github.com/user-attachments/files/21987246/Teli.platoniak.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-BamBu, 2025-08-26:
@HegyLako-eredeti Thank you for your feedback. When multiple models overlap, it will result in an error. However, you can combine these models into a single group and then print them. As shown in the GIF below; does this meet your expectations?
[image]

Haidiye00, 2025-09-01:
[image]
Hello, you can follow the instructions in the picture to achieve your goal.
Step 1: Select all the objects you want to merge and right-click to merge.
Step 2: Due to the presence of non popular edges in your model, right-click to fix them.
Step 3: First, drag the object you want to place separately to a suspended position using the translation tool.
Step 4: Using the assembly tool, first select the upper surface of the table, then select the lower surface of the triangular pyramid, and use center alignment to complete the process.