bambulab/BambuStudio #8000: 2.2.1 missing AppImage

spacemanspiff2007, 2025-08-26:
### Bambu Studio Version

2.2.1

### Where is the application from?

Bambu Lab Official website

### OS version

Linux

### Additional system information

[Release 02.02.01](https://github.com/bambulab/BambuStudio/releases/tag/v02.02.01.58) is missing the AppImage Artifacts for running under Linux.
In [02.02.00](https://github.com/bambulab/BambuStudio/releases/tag/v02.02.00.85) they are there as expected.

### Printer

-

### How to reproduce

Open Releases page

### Actual results

No AppImage

### Expected results

AppImage available

### Project file & Debug log uploads

-

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Canaill51, 2025-08-27:
> ### Bambu Studio Version
> 2.2.1
> 
> ### Where is the application from?
> Bambu Lab Official website
> 
> ### OS version
> Linux
> 
> ### Additional system information
> [Release 02.02.01](https://github.com/bambulab/BambuStudio/releases/tag/v02.02.01.58) is missing the AppImage Artifacts for running under Linux.
> 
> <img alt="Image" width="545" height="336" src="https://private-user-images.githubusercontent.com/10754716/482240375-439a5ac5-166d-45f1-9851-be72b16448ff.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTYzMjMxNjMsIm5iZiI6MTc1NjMyMjg2MywicGF0aCI6Ii8xMDc1NDcxNi80ODIyNDAzNzUtNDM5YTVhYzUtMTY2ZC00NWYxLTk4NTEtYmU3MmIxNjQ0OGZmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA4MjclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwODI3VDE5Mjc0M1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWZiNDE5YjcwYTkzNzYxMjdmNTU5NjNiNTJlOTZhMzk0MDIzZGYwMDFjNzA1MzExNmYxOTllNjE4YjYwMjFlOWMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.qXCZ5S9Hr5BVQ5UULl2dEfUvlF4OX3dYVubfcDOj-H0">
> In [02.02.00](https://github.com/bambulab/BambuStudio/releases/tag/v02.02.00.85) they are there as expected.
> 
> ### Printer
> 
> ### How to reproduce
> Open Releases page
> 
> ### Actual results
> No AppImage
> 
> ### Expected results
> AppImage available
> 
> ### Project file & Debug log uploads
> 
> ### Checklist of files to include
> * [ ]  Log file[ ]  Project file

You can use its Flatpak version, available from yesterday.

BoBoYu-BamBu, 2025-08-29:
@spacemanspiff2007 The latest Linux package has been released. Please update your usage.
[image]

closed 2025-08-29 by BoBoYu-BamBu