bambulab/BambuStudio #8003: Bambu Studio 2.2.0.85 Linux Ubuntu 24.04.3

BrettRogersUK, 2025-08-26:
### Bambu Studio Version

2.2.0.85

### Where is the application from?

Built from source by yourself or others

### OS version

Ubuntu Linux 24.04.3

### Additional system information

OS: Ubuntu Cinnamon 24.04.3 LTS x86_ 
Host: HP Laptop 15s-fq5xxx 
Kernel: 6.14.0-28-generic 
Shell: bash 5.2.21 
Resolution: 1920x1080 
DE: Cinnamon 6.0.4 
CPU: 12th Gen Intel i7-1255U (12) @ 4.700GHz  
GPU: Intel Alder Lake-UP3 GT2 [Iris Xe Graphics] 

Installed from FLATPAK (https://flathub.org/apps/com.bambulab.BambuStudio):
flatpak install flathub com.bambulab.BambuStudio

System hangs when loading Bambu Studio showing a 'X is not responding' with FORCE CLOSE and WAIT buttons. If I click on WAIT then after about 30 secs the program loads but I am unable to do anything for anohter 30 secs or so then its functional again. It happens when loading from the applications menu or when loading *.stl or *.3mf files also.

[image]

### Printer

Bambu Lab A1

### How to reproduce

1. Find Bambu Studio in the Applications Menu (Menu -> Graphics -> BambuStudio)
2. Click on Icon and Load Bambu Studio
3. OR Double Click on *.stl or *.3mf File.

### Actual results

Brings up window with FORCE CLOSE and WAIT buttons.

[image]

### Expected results

Bambu Studio Loads without issue.

### Project file & Debug log uploads

[debug_Tue_Aug_26_20_43_45_2.log.0.zip](https://github.com/user-attachments/files/21994581/debug_Tue_Aug_26_20_43_45_2.log.0.zip)

[Sweet_Box_v3.3mf.zip](https://github.com/user-attachments/files/21994593/Sweet_Box_v3.3mf.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

spacemanspiff2007, 2025-08-27:
I'm experiencing the same issue. It was introduced in 2.2.0 and was not there before.

BrettRogersUK, 2025-08-27:
> I'm experiencing the same issue with the AppImage. It was introduced in 2.2.0 and was not there before.

No, I only noticed it yesterday when I went to use it. Is it fixed in 2.2.1?? The flathub hasn't been updated yet with the new version. So I dont know and from what I saw there was no appimage available for the new version either in the releases.

BrettRogersUK, 2025-08-28:
Any idea whats causing this at all?? it is really annoying!

CryptorClub, 2025-09-01:
> Can't run AppImage s 7829 and 8017 Error in terminal `The program 'bambu-studio' received an X Window System error. This probably reflects a bug in the program. The error was 'BadValue (integer parameter out of range for operation)'. (Details: serial 385 error_code 2 request_code 149 (GLX) minor_code 3) (Note to programmers: normally, X errors are reported asynchronously; that is, you will receive the error a while after causing it. To debug your program, run it with the GDK_SYNCHRONIZE environment variable to change this behavior. You can then get a meaningful backtrace from your debugger if you break on the gdk_x_error() function.)`

My case could be partially resolved with switching off Nvidia drivers to Xorg for this past two versions.

spacemanspiff2007, 2025-09-05:
This is what I see when I open it which is essentially the same wait/force quit dialog.

[image]

The duration and amount of times I have to click wait depends.
It seems there is some blocking background communication made which sometimes take longer.
Once the application has loaded which takes about 30 - 90 secs everything runs fine.

mradeschnig, 2025-09-06:
> Delete all configuration files? On Linux, configuration file conflicts often cause abnormal behavior. For AppImage, they are located in ~/.config/BambuStudio, and for Flatpak, in ~/.var/app/com.bambulab.BambuStudio/config/BambuStudio.

For me this fixed the problem.

BrettRogersUK, 2025-09-06:
> > This is what I see when I open it which is essentially the same wait/force quit dialog.
> > <img alt="Image" width="392" height="152" src="https://private-user-images.githubusercontent.com/10754716/485961738-83b175d8-de6e-4c27-a155-e1368d371ddb.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTcwNTQ3MDIsIm5iZiI6MTc1NzA1NDQwMiwicGF0aCI6Ii8xMDc1NDcxNi80ODU5NjE3MzgtODNiMTc1ZDgtZGU2ZS00YzI3LWExNTUtZTEzNjhkMzcxZGRiLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA5MDUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwOTA1VDA2NDAwMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTU1NTI2ODMyOGY3MmEyNWVjYmNiZjZkNTEzMzUyZDc4OTJhYTViNTBjMjhlZmE5YjQ3MWFmMDM0MTgxN2FiNDcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.MfP3Yoehp5pYAzM6BmA7LqLu7N9mNIbd4siomFAU8Y8">
> > The duration and amount of times I have to click wait depends. It seems there is some blocking background communication made which sometimes take longer. Once the application has loaded which takes about 30 - 90 secs everything runs fine.
> 
> Delete all configuration files? On Linux, configuration file conflicts often cause abnormal behavior. For AppImage, they are located in ~/.config/BambuStudio, and for Flatpak, in ~/.var/app/com.bambulab.BambuStudio/config/BambuStudio.

Tried renaming the config files to .old and loaded bambustudio up again, still got the box up with force close or wait buttons. New config files were created though. But still getting box up on screen when loading Bambu Studio. Next thing to try??

CryptorClub, 2025-09-13:
> I am having the same issue with Mint 22.1 Cinnamon. The I really don't want to revert to a Windows environment just to use the printer. I've tried installing nvidia driver 550, also removed BambuStudio flatpak and tried the AppImage, worked once and then showed the wait/force quit popup. The printer connects to the iPhone app, has wifi ip, and functioning properly, except now I can't use it.
> 
> Please expedite the investigation into this issue.

Two last versions of AppImage don't work with NVIDIA drivers at my Ubuntu. I'm using Xorg now to use latest AppImage

MACKADRIS, 2025-09-14:
I am printing now ( I hope as the job went to the printer.)
I uninstalled and deleted EVERYTHING Bambu.  restart.  
Updated EVERYTHING.
Upgraded to Mint 22.2 Zara  ( not too involved)
Updated everything.
Installed Bambu Studio from Flathub
restart
Update everything
Open the Bambu Studio from the main Cinnamon menu.

The key to all this is the extreme lag in the app.  I also had an issue with saved profiles being downloaded from by Bambu online backup.  Device type was calling P1S, while I am connected to P1P
The setup popup shows, and just followed along SLOWLY, patiently waiting until a response from the last button push.
The network plugin installs, halts at 50% for a minute, then completes.  Just wait.

Everything takes forever, but it is testing bed leveling as I type this..

I bet the slow speed is because of all the errors.  Try running Bambu Studio from the terminal and see for yourself.
I think is has something to do with the libxapp-gtk3-module version conflict between 2.8.8 and the previous version that this app still looks for.

Good luck keeping customers, Bambu  !!!!  This is crap that you are not supporting properly.

BrettRogersUK, 2025-10-03:
@MackBambu any progress on sorting this bug?? It seems more and more people are being affected by it and quite frankly it's very poor customer service from Bambu giving us sub-standard software to use that doesn't work properly.

johnnytolengo, 2025-10-23:
Seems to be the network_module:

[image]

seeraven, 2025-10-29:
> I compiled BambuStudio-02.03.00.70 with set(SLIC3R_VERSION "02.00.01.01") and set(SLIC3R_VERSION "02.01.01.52") It starts up with no hang and sliced my part fine. But when I try to print I get the Bambu Studio error Exception::bad_alloc.
> 
> Edit: Since switching network plugin didn't help, is there a way to compile so that it behaves like it did on my old machine? I don't know what that would actually mean, but it did work on that machine, so it is machine dependent.

That would involve tracing down all changes that interact with the usage of the network plugin and modifying them to work with the old plugin. That can be done, but is a major effort. And since the plugin is closed source from BambuLab, we don't know what changed, why they changed it, and how to work around it (like playing new network plugin but in Bambu Studio).

The main problem I see here is that BambuLab is not responding at all. I just opened a support ticket and hope to get more attention on this bug.

BrettRogersUK, 2025-10-30:
@MackBambu Yet Again another release and still the same issue as the previous 4 releases since this BUG was introduced. It does it also with the APPIMAGE as well as the FLATPAK. WHY IS NOBODY DOING ANYTHING ABOUT THIS!!?!?!?

@BoBoYu-BamBu

MACKADRIS, 2026-01-01:
At this time, I am able to open BambuStudio.  It still shows the Wait or Force Stop popup.  After that, I wait a couple minutes, and then just import, arrange, add whatever supports, changes, etc, and it prints without issue.  I am running Linux Mint 22.2, and have not had any issues with configuring or printing to my P1P.  It takes some time to load, but works.

If anyone cannot print, the only difference may be that I installed the Orca print app, but never used it.  I opened it ONCE.  Everything has worked, except for the time wait at the beginning.  I am not sure if the network plugin has anything to do with the slow loading, but I do know that if you run the app in Linux using the terminal, it does show errors.