bambulab/BambuStudio #8005: Cannot manually calibrate filament in P1S with AMS

jfxg, 2025-08-26:
### Bambu Studio Version

2.2.1.58

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

### Printer

P1S

### How to reproduce

1. Go to Calibration
2. Select either Flow Dynamics or Flow Rate (both exhibit problem)
3. Click "Manual Calibration"
4. Select options, choose filament in AMS
5. Click "Calibrate"

### Actual results

Printer starts warming up to print the calibration patterns, but then says the filament has run out. It never attempts to load filament from the AMS. The error window telling me to load more filament appears to be giving instructions for non-AMS filament loading, illustrating the inlet on the back of the printer.

### Expected results

I expected it to load the selected filament from the AMS for calibration. I'm nearly certain I've done this successfully in the past for other custom filaments.

### Project file & Debug log uploads

[debug_Tue_Aug_26_16_19_03_49788.log.zip](https://github.com/user-attachments/files/21998191/debug_Tue_Aug_26_16_19_03_49788.log.zip)

[failed_calibration.zip](https://github.com/user-attachments/files/21998195/failed_calibration.zip)

[Screenshots.zip](https://github.com/user-attachments/files/21998200/Screenshots.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

jfxg, 2025-08-26:
I continued experimenting with this. If I first load the filament to the toolhead manually from the Device page (still using the AMS), and then start the calibration while it is currently loaded, it _still_ gives me the filament ran out error, but I can click "Filament Loaded, Resume" at this point and it continues with the calibration. That does not work when i start calibration with the filament unloaded as it repeatedly shows the error and never attempts to load from the AMS.

FantomeXS, 2025-08-27:
Same issue here. Exactly the same setup with P1S and AMS. Cannot do calibration through Bambu Studio. P1S prints fine, AMS loads/unloads fine. Only issue is on the Flow Dynamics Calibration. Tried both line and pattern, neither worked.