bambulab/BambuStudio #8011: (02.02.01.58) Calibration (flow Dynamics) failed to load filament from AMS-HT

michael2schrader-cpu, 2025-08-27:
### Bambu Studio Version

02.02.01.58

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS Sequoia 15.6.1

### Additional system information

iMac Apple M1, 24“, 2021, 16GB memory, 1TB HD

### Printer

Bambu P1S (1x AMS2 Pro, 2xx AMS-HT)

### How to reproduce

1. Calibration —> Flow Dynamics
2. Manual calibration
3. Nozzle 0.4, plate: textured PEI, AMS-HT-1 (A): Generic PETG, Method: Line, From 0.01 to 0.06, Step 0.002
4. Hit „calibrate“
5. Studio is sending request to printer, filament is not been loaded
6. After a few seconds error-message: "Filament ran out. Please load new filament. (0300-8004-205927)

### Actual results

After error-message nothing happen. Hit „Load filament“ —> nothing, Hit „Filament loaded, Resume“ —> nothing.

Deleted BS 02.02.01.58 and install 02.01.01.52 again.
(BS 02.02.00.85 not installed, has problem to select filament type under „Device“ for the AMS lite (Bambu A1))

BoBoYu-BamBu, 2025-08-28:
@michael2schrader-cpu Thank you for your feedback. We are addressing this issue. If you urgently need calibration, you can use the old version first.

BoBoYu-BamBu, 2025-08-28:
@michael2schrader-cpu  This issue has been resolved. Please try again using the latest version V2.2.1.60. If you have any problems, please feel free to let us know. Download link:https://github.com/bambulab/BambuStudio/releases

closed 2025-08-29 by BoBoYu-BamBu