bambulab/BambuStudio #8026: Prepare panel too narrow

baschwar, 2025-08-28:
### Bambu Studio Version

2.2.1.60

### Where is the application from?

Bambu Lab Official website

### OS version

Mac OS Sequoia 15.6.1

### Additional system information

[image]

### Printer

Bambu X1c

### How to reproduce

1. Open app
2. Default should open to prepare tab 

### Actual results

Tab is too narrow and not adjustable. 

### Expected results

Proper width tab so all elements can be seen without cutting text and fields off.

### Project file & Debug log uploads

n/a

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-BamBu, 2025-08-29:
@baschwar If there are still any issues, could you please provide the configuration file (.conf)? You can send it to my email address yuting.huang@bambulab.com? As shown in the following figure:
[image]

Haidiye00, 2025-08-29:
Hello, please follow the suggestion mentioned above to conduct testing and confirmation. Thank you