bambulab/BambuStudio #8043: Changing filament temperature does not actually change the printing temp

SteveDC2, 2025-08-29:
### Bambu Studio Version

2.2.1.58

### Where is the application from?

Bambu Lab Official website

### OS version

WIndows 11

### Printer

H2D

### How to reproduce

Click "Select Filament to Use" in left panel to open filaments.
Click "Custom Filaments"
Click "Create New"
Check "Can't find vendor" & enter vendor name, e.g. "Spectrum"
Select Type = "PLA"
Enter serial, e.g. "UL94"
Select "Copy current filament preset"
Check "Bambu H2D"
Select "Bambu PLA Basic"
Click "Create" then "OK", then "Confirm"
Select "Prepare" tab
Click dropdown for filament1 and select previously created filament.
Click filament 1 "click to edit' button
Change Print temperature - nozzle - Initial layer and other layers to 200
Click save icon
Click "X" to close the dialog
Load an STL model
Click the object to select it
Press "1" to select filament 1 for the object
Click "Slice Plate"
If Syn Now appears click "Later"
In slicing result display option dropdown select "Temperature" to display print temperature range
Note that the temperature is 220 and NOT 200 as expected.
Exporting the GCode and examining also shows the nozzle temperature is 220, and printing is done at 220

### Actual results

Baseline for filament creation...

[image]

Filament selection and assignment to the model...

[image]

Resultant printing temperature review...

[image]

### Expected results

Expect print temperature to be 200

### Project file & Debug log uploads

[TemperatureIssue.zip](https://github.com/user-attachments/files/22054162/TemperatureIssue.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

rouyuanzi-bambu, 2025-09-08:
Hello, thank you for your feedback. Since the H2D nozzle has both Standard and High Flow modes, in order to further assist us in troubleshooting, could you please confirm whether the nozzle temperature you modified matches the mode used during slicing? For example, if you modified the temperature for the High Flow nozzle, please ensure that slicing was also performed in High Flow mode.[image]
slicing:
[image]

ShanCang-BamBu, 2025-09-08:
Hello,
The JSON file contains two values: the filament parameters for Standard Flow Mode and High Flow Mode.
When you change the nozzle temperature in Standard Flow Mode and then switch to High Flow Mode, the parameter remains at its original value.

[image]

[image]

Therefore, if you need to modify the filament parameters for a specific mode, please make sure that both the parameter settings and the slicing settings are configured for the correct print mode.

[image]

[image]

SteveDC2, 2025-09-10:
Ahh. OK. Makes sense now I guess. Thanks for the explanation.

Minor suggestion. Would be good if the temperature selection (or any other nozzle dependent parameter) automatically selected the current setting since I would expect most people would be changing the currently active configuration.

Anyhow, thanks for the explanation.

BR,

Steve