bambulab/BambuStudio #8049: 侧边栏突然变窄且无法调整

dr4lamb, 2025-08-30:
### Bambu Studio Version

2.2.1.60

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 15.6.1

### Printer

p1sc

### How to reproduce

不知道为什么突然出现

### Actual results

侧边栏突然变得很窄，回退版本，重装App均无效。
尝试按照 https://github.com/bambulab/BambuStudio/issues/7377 里面的指引修改BambooStudio.conf中的enable_sidebar_resizable为true，重启应用后依然无法调整侧边栏。

### Expected results

侧边栏正常，或者可以手动调节

### Project file & Debug log uploads

[debug_network_Sat_Aug_30_12_06_46.log.enc.zip](https://github.com/user-attachments/files/22056643/debug_network_Sat_Aug_30_12_06_46.log.enc.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

dr4lamb, 2025-08-30:
把资源库中的BambuStudio整体删除，然后重启应用后侧边栏正常了，并且也可以调整。不知道是否是因为旧版本中的某些配置信息有干扰，建议官方组可以调查一下。

Haidiye00, 2025-09-01:
Yes, your operation method is correct. It was indeed affected by the old conf file, but we have tried to ignore the old impact once. There may still be a few cases that have not been considered. We are very sorry

Haidiye00, 2025-10-16:
Hello, there are no other questions. This order is temporarily closed. If there are any issues, we will continue to discuss~

closed 2025-11-03 by Haidiye00