bambulab/BambuStudio #8051: Support H2D with different nozzle sizes on the left and right

AnalogKnight, 2025-08-30:
The H2D has been released for over 4 months. Initially, users were told that further uses for the dual nozzles were still under development, but so far, nothing has been updated. It seems the only benefit of the dual nozzles is reducing filament waste. Now that the H2S has been released, it's priced 30-40% lower than the H2D. If we used that money on filament, we could buy at least 50 spools—the flushing waste is negligible.

The benefits of using two nozzles of different sizes are obvious: we can use a 0.2mm nozzle to print exterior walls and a larger nozzle for infill. This can save a lot of time while still producing high-quality models.

Are there plans to add more functionality to the H2D's dual nozzles beyond simply saving filament?

malventano, 2025-08-30:
You can still use wider line widths for inner walls. Applies more to 0.4>0.6 in terms of speed, but you can still easily do 0.4 width for infill on a 0.2, with the bonus that you don't even need to switch extruders.

AnalogKnight, 2025-08-31:
> You can still use wider line widths for inner walls. Applies more to 0.4>0.6 in terms of speed, but you can still easily do 0.4 width for infill on a 0.2, with the bonus that you don't even need to switch extruders.
> 
> More here: https://www.cnckitchen.com/blog/the-effect-of-extrusion-width-on-strength-and-quality-of-3d-prints

Thanks for the info, but I don't think it solves the awkward situation with the H2D's dual nozzles. Yes, I can print 0.4mm with a 0.2mm nozzle. But I can also print 0.6mm or higher with a 0.4mm nozzle, and the same goes for a 0.6mm or 0.8mm nozzle. Allowing for different nozzle sizes is always a benefit.

iamniels, 2025-09-02:
I agree with this issue. One application example is printing text with a 0.2mm nozzle on a larger object.

abudden, 2025-10-28:
> I agree with this issue. One application example is printing text with a 0.2mm nozzle on a larger object.

This would be my primary use for mixed nozzles.  Small text on top of a large object is currently impractical as the small nozzle would take too long and the large nozzle can't print text in a fine-enough resolution.

For me, I'd be perfectly happy if it had to be layer-by-layer (i.e. any specific layer could only be printed with a single nozzle size but you could do the last couple of layers in a different nozzle size to the previous ones).  This would allow big things to be printed and then small text to be written on top.

nineimer, 2026-08-24:
I would like to add an engineering and purchasing perspective to this feature request, particularly regarding the H2C and Vortek system.

I am a mechanical engineer currently considering which Bambu Lab printer to buy, mainly for engineering prototypes and functional parts.

For me, support for different nozzle diameters within the same print job would be one of the strongest reasons to choose the H2C specifically.

A practical use case would be:

0.2 mm nozzle for detailed external walls, lettering, small holes and fine geometric features
0.4 mm nozzle for normal structural walls and general geometry
0.6 or 0.8 mm nozzle for fast internal infill and large-volume deposition

This could combine fine external quality with significantly shorter print times.

At the moment, if one area of a part requires the resolution of a 0.2 mm nozzle, the user may have to accept the print-time penalty of using that nozzle for much more of the model than actually requires it.

With Vortek, the hardware already has the ability to automatically exchange hotends. That creates the possibility of using the most appropriate nozzle for each type of geometry, instead of forcing the entire print job to conform to one nozzle diameter.

The layer-height challenge looks manageable

I understand that this is not simply a matter of assigning different nozzles to different features. Mixed nozzle diameters create additional slicing, calibration and print-quality challenges.

However, I do not see the layer-height problem as a fundamental limitation.

An initial implementation could restrict mixed-nozzle operation to compatible layer schedules.

For example:

0.2 mm nozzle at 0.10 mm layer height
0.6 mm nozzle depositing infill at 0.30 mm intervals

The fine nozzle could print at:

0.10 → 0.20 → 0.30 → 0.40 → 0.50 → 0.60 mm

while the coarse nozzle deposits accumulated infill at:

0.30 → 0.60 mm

In other words, the slicer could synchronize the different tools at compatible Z positions.

This is conceptually similar to existing slicing strategies where infill can be combined over several perimeter layers.

Of course, a robust implementation would also need to account for:

nozzle X/Y/Z offsets
hotend calibration
maximum layer height
extrusion width
wall/infill interface quality
collision avoidance
tool-change behavior
material compatibility

That is clearly a non-trivial software task, but it seems to me to be an advanced slicing problem rather than a fundamental hardware limitation.

Keep the normal Bambu experience simple

I do not think this functionality should make Bambu Studio more complicated for ordinary users.

A possible solution would be to separate it into two levels:

Standard Mode

The current Bambu experience:

simple
automatically validated
one compatible nozzle diameter per print
minimal configuration required
Advanced Mixed-Nozzle Mode

An explicitly enabled mode for experienced users, where different nozzles can be assigned to specific features, for example:

External perimeter → 0.2 mm
Internal perimeter → 0.4 mm
Sparse infill → 0.6 mm
Solid infill → 0.4 or 0.6 mm
Top surface → 0.2 or 0.4 mm
Support → 0.6 mm
Support interface → 0.2 or 0.4 mm

Bambu Studio could initially reject combinations that cannot produce a compatible layer schedule instead of trying to support every possible combination from day one.

Later, this could potentially evolve into an automatic optimization mode such as:

Quality / Balanced / Speed

where Bambu Studio itself selects the most appropriate available Vortek nozzle for each feature.

Why this matters commercially

From my point of view, this changes the value proposition of the H2C.

Without this capability, I can buy a substantially less expensive Bambu printer and still produce excellent prototypes.

With intelligent mixed-diameter Vortek printing, the H2C becomes something fundamentally different:

one machine capable of combining fine-detail printing and high-volume deposition automatically within the same component.

That is a capability I would be willing to pay considerably more for.

I am currently evaluating which Bambu Lab printer to purchase, and support for mixed-diameter Vortek printing would materially influence my decision toward the H2C.

I strongly suspect I am not the only engineering or prototyping customer for whom this feature could turn Vortek from an interesting tool-changing mechanism into a compelling reason to buy the printer.

The hardware is already there. I would very much like to see Bambu Studio unlock its full potential.

rogeriocamorim, 2026-10-01:
H2C owner here, with the same request and some test data.

I built a local Bambu Studio that assigns a nozzle diameter per filament, with height bands. On the H2C this already works for **left ≠ right** diameters, and for **one extra diameter on the Vortek rack** (for example 0.8 left, 0.2 → 0.4 on the rack).

What doesn't work is a **second change diameter on the rack** (0.2 → 0.6 → 0.4). The firmware validates every mid-print hotend change against the single `nozzle_diameter` per extruder in the job header, and stops with "Hotend type mismatch… slot X" even though the hotend is correct. That happens even when the job is sent from the official Bambu Studio with a correct per-filament nozzle mapping.

Full test matrix, photos and a proposed fix: #12431. If you want rack mixing on the H2C, a 👍 there would help.

heinemil, 2026-10-30:
+1 to OP. Combining fine detail with faster bulk printing should be a priority for a premium dual-nozzle printer.

A 0.2 mm nozzle for detailed walls and text, alongside a 0.6–0.8 mm nozzle for infill, would make the H2D substantially more useful. Increasing line width on a small nozzle is not an equivalent solution.

OrcaSlicer 2.5.0 alpha already supports mixed nozzle sizes on the H2D and X2D:
https://github.com/OrcaSlicer/OrcaSlicer/releases/tag/v2.5.0-alpha

Why should customers need third-party software to unlock this capability on Bambu’s own hardware?

This request has been open for over a year. Please provide a concrete update: is this actively being developed, what is blocking it, and what release window are you targeting? An opt-in experimental mode would already be a meaningful step.

This directly affects the value of buying a dual-nozzle printer. Bambu Studio should help justify that premium—not be the reason customers cannot fully benefit from it.