bambulab/BambuStudio #8055: FEATURE REQUEST: Add Global Speed Factor Control in Studio

sealyons, 2025-08-30:
FEATURE REQUEST: Add Global Speed Factor Control in Studio

SUMMARY:
Introduce a Speed Factor (%) parameter in Bambu Studio that allows users to scale all print speeds globally (or within modifiers such as Height Range Modifiers), similar to how Flow/Extrusion Multipliers work.

PROBLEM:
Currently, when tuning print speeds in Studio (especially with multi-filament AMS prints), users must manually adjust each speed parameter (outer wall, inner wall, infill, top surface, etc.) within the Process Profile.
This is time consuming and error-prone.

When using Height Range Modifiers for multi-filament AMS prints, it is not possible to quickly adjust overall speed across a region — you must tweak individual speed settings one by one.

This makes balancing mixed filament types (e.g., high-speed Matte PLA + regular Matte PLA + specialty PLAGRAD) much harder than necessary.

PROPOSED SOLUTION:
Add a Speed Factor (%) field at both the global level and within Height Range Modifiers.

_Example:_ Setting Speed Factor to 90% would automatically reduce all speed parameters (outer/inner wall, infill, supports, top surface, etc.) by 10%. Conversely, setting Speed Factor to 110% would raise them all by 10%.

Works in combination with filament profile MVS limits, so users cannot exceed safe extrusion limits.

BENEFITS:
* Simplifies workflow for users managing multi-filament AMS prints where different filaments tolerate different speeds.
* Greatly speeds up testing and iteration — instead of editing 10+ speed fields, one slider adjusts them all.
* Makes Height Range Modifiers far more powerful for tuning complex prints (slowing down decorative upper layers, speeding up structural bases, balancing mixed filament sections).

Gask44, 2026-09-05:
I would also really like to see a 70–75% speed mode added between Silent (50%) and Standard (100%).

This would be especially useful when printing ASA, ABS and other engineering filaments.

For example, I am currently printing a large multi-part ASA project on a Bambu Lab H2S. At 100% speed I started seeing some corner warping, so I switched to Silent mode at 50% to give the print more stability. It helped, but the jump from 100% to 50% is simply too large — the remaining print time increased dramatically.

A 75% mode would be a very useful middle ground: slower and gentler than Standard mode when extra print stability is needed, without almost doubling the print time by switching to Silent mode.

Something like:

Silent — 50%  
Stable / Quality — 75%  
Standard — 100%  
Sport — 124%  
Ludicrous — 166%

This would be particularly valuable for ASA, ABS, PC and other engineering materials, as well as large prints where reducing speed can help once signs of warping or instability appear during an ongoing print.

It would be even better if 75% could be selected directly from the printer, Bambu Studio and Bambu Handy just like the existing speed modes.

kcahrens, 2026-11-09:
yes please it's dumb that every model I download I have to re- configure the speeds, most of them get set to the max of the A1 then I have to adjust. why not just have a feature save a profile or set default / default range for your device.