bambulab/BambuStudio #8060: Arabic text

honeydashtii, 2025-08-31:
Hello,  
I would like to request support for Arabic language in Bambu Studio.  
Currently, when I try to add Arabic text, it appears broken because the software does not handle Right-to-Left (RTL) text correctly.  

Arabic is spoken by more than 400 million people worldwide, and supporting it would be very helpful for many users.  

Thank you for considering this feature request.

penanot, 2025-09-02:
> Hello, I would like to request support for Arabic language in Bambu Studio. Currently, when I try to add Arabic text, it appears broken because the software does not handle Right-to-Left (RTL) text [...]

It could be a nice feature to add on.

alkobra1394, 2025-09-06:
I would like to request support for Arabic language in Bambu Studio