bambulab/BambuStudio #8069: 切片层数错误

cike-567, 2025-09-01:
### Bambu Studio Version

2.2.1.60

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11 25H2（26200.5074）

### Additional system information

CPU：R9 9950X
内存：64G
GPU：RX 6700XT

### Printer

Bambu lab A1

### How to reproduce

1. 导入格式为 “.stp” 的模型。
2. 点击 “切片单盘”。

### Actual results

图中的孔洞，层数不均匀，上层为 5 层，中层为 6 层，下层为 4 层。

[image]

### Expected results

层高设置为 0.2，图中的孔洞的层数应均匀分布，上、中、下各 5 层。

### Project file & Debug log uploads

[Desktop.zip](https://github.com/user-attachments/files/22080746/Desktop.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

Diatom-Banbu, 2025-09-03:
感谢您的反馈，经过排查，这种层的划分应该是正常现象，并非bug。
这种精度舍入是FDM打印常见的层高引起的误差，您的模型总高度20mm，从底面到槽孔下表面总计8.5mm。对于层高0.2mm来说，刚好在一半层高处开始变为槽孔。层生成的高度依此从0-0.2, 0.2-0.4……直到8.4-8.6，此时8.5mm的层高刚好处于此层中间高度，这里的生成结果实际上将其视为完整的一层（多生成了半层0.1mm），同时顶层也就是11.5mm高度的时候是槽孔顶部少生成半层，因此总的孔宽不变但向上了0.1mm。这种策略是由于0.2mm的层高无法产生0.1mm的高度划分，因此必须**将孔近似地分割为0.2mm一层**的走线。
一种验证该问题的方法是，将您的模型切割掉一部分（例如8.5mm，或者0.1mm，只要是0.2mm的整数倍多0.1mm即可）
[image]
再次切片，会发现孔的分层由于对齐了层高，变得符合均匀分布，上、中、下各 5 层。

[image]

如果您希望孔具有更高的精度，有以下两种方法您可以参考：
1. 修改设计，使孔的位置距离底面具有整数倍数的层高，从而使切片精准（例如将总高度改为19.8，3mm宽度槽孔在中间，那么8.5mm变为8.4mm）；
2. 修改层高，例如使用0.12mm层高进行打印，这种误差将被减少（上8，中9，下8），但这会减慢打印速度。

[image]

cike-567, 2025-09-04:
谢谢你的回答。