bambulab/BambuStudio #8075: Cannot install file on Linux: this doesn't look like a squashfs file

aqwek, 2025-09-02:
### Bambu Studio Version

02.02.1.60

### Where is the application from?

Bambu Lab github releases

### OS version

Arch Linux 6.16.4-arch1-1

### Printer

none

### How to reproduce

1. Either download the AppImage from this repo, or install from the AUR using yay/paru -S bambustudio-bin

### Actual results

Running the AppImage and installing the package will fail with error "This doesn't look like a squashfs image."

yay error:
==> Starting package()...
qt.qpa.plugin: Could not find the Qt platform plugin "wayland" in ""
This doesn't look like a squashfs image.
Failed to open squashfs image
==> ERROR: A failure occurred in package().
    Aborting...
 -> error making: bambustudio-bin-exit status 4
 -> Failed to install the following packages. Manual intervention is required:
bambustudio-bin - exit status 4

AppImage error:
./Bambu_Studio_linux_fedora-v02.02.01.60_71f45d48cf8f4cfe67b18a2ac2522d69.AppImage 
qt.qpa.plugin: Could not find the Qt platform plugin "wayland" in ""
fuse: memory allocation failed
squashfuse 0.5.2 (c) 2012 Dave Vasilevsky

Usage: squashfuse [options] ARCHIVE MOUNTPOINT

(null) options:
    -o offset=N            offset N bytes into ARCHIVE to mount
    -o subdir=PATH         mount subdirectory PATH of ARCHIVE
    -o notify_pipe=PATH    named pipe that will receive 's' (success)
                           or 'f' (failure) when the mountpoint is ready
    -o timeout=N           idle N seconds for automatic unmount
    -o uid=N               set file owner to uid N
    -o gid=N               set file group to gid N

FUSE options:
    -h   --help            print help
    -V   --version         print version
    -d   -o debug          enable debug output (implies -f)
    -f                     foreground operation
    -s                     disable multi-threaded operation
    -o clone_fd            use separate fuse device fd for each thread
                           (may improve performance)
    -o max_idle_threads    the maximum number of idle worker threads
                           allowed (default: -1)
    -o max_threads         the maximum number of worker threads
                           allowed (default: 10)
fuse: memory allocation failed
squashfuse 0.5.2 (c) 2012 Dave Vasilevsky

Usage: squashfuse [options] ARCHIVE MOUNTPOINT

(null) options:
    -o offset=N            offset N bytes into ARCHIVE to mount
    -o subdir=PATH         mount subdirectory PATH of ARCHIVE
    -o notify_pipe=PATH    named pipe that will receive 's' (success)
                           or 'f' (failure) when the mountpoint is ready
    -o timeout=N           idle N seconds for automatic unmount
    -o uid=N               set file owner to uid N
    -o gid=N               set file group to gid N

FUSE options:
    -h   --help            print help
    -V   --version         print version
    -d   -o debug          enable debug output (implies -f)
    -f                     foreground operation
    -s                     disable multi-threaded operation
    -o clone_fd            use separate fuse device fd for each thread
                           (may improve performance)
    -o max_idle_threads    the maximum number of idle worker threads
                           allowed (default: -1)
    -o max_threads         the maximum number of worker threads
                           allowed (default: 10)
squashfuse 0.5.2 (c) 2012 Dave Vasilevsky

Usage: squashfuse [options] ARCHIVE MOUNTPOINT

(null) options:
    -o offset=N            offset N bytes into ARCHIVE to mount
    -o subdir=PATH         mount subdirectory PATH of ARCHIVE
    -o notify_pipe=PATH    named pipe that will receive 's' (success)
                           or 'f' (failure) when the mountpoint is ready
    -o timeout=N           idle N seconds for automatic unmount
    -o uid=N               set file owner to uid N
    -o gid=N               set file group to gid N

FUSE options:
    -h   --help            print help
    -V   --version         print version
    -d   -o debug          enable debug output (implies -f)
    -f                     foreground operation
    -s                     disable multi-threaded operation
    -o clone_fd            use separate fuse device fd for each thread
                           (may improve performance)
    -o max_idle_threads    the maximum number of idle worker threads
                           allowed (default: -1)
    -o max_threads         the maximum number of worker threads
                           allowed (default: 10)
Can't open squashfs image: Bad address

Cannot mount AppImage, please check your FUSE setup.
You might still be able to extract the contents of this AppImage 
if you run it with the --appimage-extract option. 
See https://github.com/AppImage/AppImageKit/wiki/FUSE 
for more information
execv error: No such file or directory

### Expected results

AppImage and package should run fine.

### Project file & Debug log uploads

Application did not start.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

mikeyoung83, 2025-09-03:
Same error on updating. Uninstalled, tried to reinstall and getting the above error.

MackBambu, 2025-09-03:
Can you try using the Ubuntu 24 version of the AppImage? It works fine for me on Arch Linux 6.9.
Or use the Flatpak version.

[image]

SpaceEnergy, 2025-09-03:
@MackBambu 
> Can you try using the Ubuntu 24 version of the AppImage? It works fine for me on Arch Linux 6.9. Or use the Flatpak version.

I use Hyprland on CachyOS, and the AppImage, Flatpak and AUR don't work.

MackBambu, 2025-09-05:
> [@MackBambu](https://github.com/MackBambu) I was just reading the comments here, from vikingowl: https://aur.archlinux.org/packages/bambustudio-bin

This package is not officially maintained, so there’s nothing that can be done.

5g6zjt8m, 2025-09-13:
I am facing this same issue as well. Verified that newly installed appimages load up fine and FUSE is functioning. Ubuntu 24.04 X11. The Flatpak version least loaded the software, but it ran poorly.

Just use OrcaSlicer until this is fixed. Had no issues getting it running.

cycl0ne, 2025-09-17:
Tried to Upgrade my 2.01 2 Days ago to the new version here: 
[bambustudio-02.02.02.56.AppImage](https://github.com/bambulab/BambuStudio/releases/download/v02.02.02.56/Bambu_Studio_ubuntu-24.04_PR-8184.AppImage)

But it fails, running from the AUR Archive:
 ==> Starting package()... This doesn't look like a squashfs image. Failed to open squashfs image ==> ERROR: A failure occurred in package(). Aborting

Starting as the AppImage directly:
fuse: memory allocation failed
Cannot mount AppImage, please check your FUSE setup.

SpaceEnergy, 2025-09-18:
> Anyone reading this: It is a problem in the AppImageLauncher2.2 -> You need to install AppImageLauncher3.0Beta .

It seems Bambu uses a newer compression type which is not detected by AppImageLauncher (they use an old version of fuse app to call the AppImage).

This helped, thanks!