bambulab/BambuStudio #8077: Problème Bambu Studio

alexisbouyon, 2025-09-02:
### Bambu Studio Version

2.2.1.60

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu lab H2S

### How to reproduce

N'importe quand

### Actual results

Lors de l'enregistrement de la modification d'un profil tout se passe bien puis à l'ouverture mon profil à disparu ainsi que les préréglages de mon imprimante

### Expected results

Correction du beug 

### Project file & Debug log uploads

[Probleme.zip](https://github.com/user-attachments/files/22091529/Probleme.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

rouyuanzi-bambu, 2025-09-03:
Bonjour, Nous vous remercions pour votre retour. Nous n'avons pas réussi à reproduire le problème que vous avez décrit dans la version 2.2.1.60.Pour nous aider à analyser la situation plus en détail, pourriez-vous s'il vous plaît nous fournir le fichier journal (log) ?
Voici la procédure pour obtenir les journaux :
[https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log](https://www.google.com/url?sa=E&q=https%3A%2F%2Fwiki.bambulab.com%2Fen%2Fx1%2Ftroubleshooting%2Fbambu-studio-log)
Merci pour votre aide.

rouyuanzi-bambu, 2025-09-03:
Merci beaucoup pour votre soutien ! Nous avons bien reçu vos commentaires et allons analyser le problème pour en identifier la cause dans les plus brefs délais.

rouyuanzi-bambu, 2025-09-03:
Bonjour, Je ne vois que deux tickets GitHub sur votre profil. La question pour laquelle vous n'avez pas reçu de réponse est-elle bien celle du lien ci-dessous ? https://github.com/bambulab/BambuStudio/issues/7888
Pourriez-vous s'il vous plaît confirmer ?
Merci !

alexisbouyon, 2025-09-03:
Bonjour non c'est un autre ticket
Je vous joint le n°US250902276001

Merci

alexisbouyon, 2025-09-03:
> Bonjour,
> Nous avons bien contacté le personnel compétent pour traiter votre ticket. Nous vous remercions de votre patience.

Merci beaucoup

closed 2025-09-04 by alexisbouyon