bambulab/BambuStudio #8081: Save new or edited Notes in Others

allanx17, 2025-09-02:
### Bambu Studio Version

2.2.0.85

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambu Lab A1

### How to reproduce

1. Edit the text in Others-Notes
2. The change is not shown in File/Save and the Save icon is still gray
3. If using Ctrl+S and exit, then the new text is lost.

### Actual results

Risk of loosing the new notes if not using Shift+Ctrl+S.

### Expected results

Editing in Others-Notes shall enable Save and the Save icon.

### Project file & Debug log uploads

[Others-Notes.zip](https://github.com/user-attachments/files/22101431/Others-Notes.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

blueberrybigboy, 2025-09-02:
I believe this fixes it.
[swag.zip](https://github.com/user-attachments/files/22101425/fix.zip)

Haidiye00, 2025-10-15:
https://github.com/bambulab/BambuStudio/releases/tag/v02.03.00.70
Hello, the new version has been released. Please update the software. Thank you~