bambulab/BambuStudio #8091: Unable to send file to Local Storage on H2S with properly formatted USB drive - "The printer does not support sending to the printer storage"

ElectrikCirkuit, 2025-09-03:
### Bambu Studio Version

2.2.1.60

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS 15.6.1 (24G90)

### Additional system information

CPU: 2.6 GHz 6-Core Intel Core i7
RAM: 32 GB 2400 MHz DDR4

### Printer

H2S

### How to reproduce

1. Insert a USB 3.1 flash drive into the H2S printer.
2. Format the USB using the on-screen format option.
3. Printer UI recognizes the USB and shows storage space.
4. In Bambu Studio 2.2.1.60, slice a model.
5. Select “Send” option from split button.
6. Modal window opens .
7. Message "The printer does not support sending to the printer storage." is displayed and "Send" button is greyed out.

I have replicated this issue with a 128 GB Samsung 3.1 flash drive, a 32 GB Sandisk 3.1 flash drive, and a 8 GB USB 2.0 flash drive. I've also been unsuccessful sending a file when testing the same version of Bambu Studio on a M2 Macbook Air. Lastly, I also tried formatting my 128 GB flash drive to ExFat using my Macbook, and I still experienced the same issue.

### Actual results

[image]

### Expected results

Send button should be enabled and allow sending of file to the printer.

### Project file & Debug log uploads

[AMS 2 Pro Dry Pots - center hygrometer.3mf.zip](https://github.com/user-attachments/files/22126952/AMS.2.Pro.Dry.Pots.-.center.hygrometer.3mf.zip)

[debug_Wed_Sep_03_15_35_26_48902.log.0.zip](https://github.com/user-attachments/files/22126983/debug_Wed_Sep_03_15_35_26_48902.log.0.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

rouyuanzi-bambu, 2025-09-04:
Hello, thank you for your feedback. This issue is caused by a corrupted H2S configuration file in Studio. The current temporary solution is to delete the printers folder in the configuration directory and then restart Studio to restore normal operation. Please try this first, and if it does not resolve the issue, feel free to contact us. We will work on fixing this problem as soon as possible.
[image]

rouyuanzi-bambu, 2025-09-05:
Hello, we are very sorry for the inconvenience caused. We have not been able to reproduce the issue you described. To help us further analyze the cause and resolve your problem, could you please provide the latest complete log files as instructed below? https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log

ElectrikCirkuit, 2025-09-05:
@rouyuanzi-bambu, attached are the logs.

[H2S_Send_Issue_Logs.zip](https://github.com/user-attachments/files/22163825/H2S_Send_Issue_Logs.zip)

rouyuanzi-bambu, 2025-09-05:
Thank you for your support. We will resolve it as soon as possible.

milk-pure, 2025-09-05:
Hello, I'd like to confirm with you two things: first, whether you clicked "Cancel" during the sending process; second, your firmware version. If it's convenient for you, could you please provide the firmware version?

milk-pure, 2025-09-09:
Hello, we are currently investigating this issue and would like to obtain more information from you. Is the prompt indicating "send failed, please retry" bound to occur? At present, we have not encountered this error prompt when using the same firmware version (01.01) as yours. Could you please provide feedback on the failure rate? Our team is still attempting to reproduce the issue.