bambulab/BambuStudio #8092: Problem with floating regions in tree supports

Ptitlion, 2025-09-03:
### Bambu Studio Version

2.2.0.85

### Where is the application from?

Bambu Lab Official website

### OS version

Mac OS 13.7.6

### Additional system information

Mac Pro

### Printer

Bambu Lab A1

### How to reproduce

Tree support generation on complex models. Always reproductible. 

### Actual results

The preview shows layer inconsistencies. Some areas are floating, which makes the print impossible. It is the same with Orca Slicer. Tree generation is so much better with Cura. 

[image]
[image]

### Expected results

Tree supports should be functional and consistent, enabling reliable prints, with improved handling of overhangs. 

### Project file & Debug log uploads

No

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Ptitlion, 2025-09-04:
Hello, unfortunately, I cannot share the 3D model due to a confidentiality agreement. To provide some additional information: the 3D model has no mesh issues, but it is very complex with many overhangs and details. It has more than one million triangles. The model itself is not the cause, since tree supports work perfectly in Cura on the same file.
The aberrations appear regardless of layer height or tree support mode. They are most significant in Strong, Auto, and Hybrid tree modes, and somewhat less in Slim mode. The aberrations are also more pronounced when tree supports are set to build plate only.
There are several types of issues: cones suddenly appearing in mid-air, illogical branch continuities extending into empty space, and triangular slice shapes also floating in mid-air.
I am attaching some additional photos for reference. Thank you.

[image]
[image]
[image]
[image]
[image]
[image]
[image]

NothFie, 2025-09-08:
@Ptitlion Thanks for your feedback. This is an old issue in Organic Tree when many trees generated in a low height. In such situation, I suggest you try the Hybrid Tree and we will fix this issue in the future.

Ptitlion, 2025-09-08:
Thank you for your reply. Hybrid mode does work somewhat better, but it still shows defects, and the trees are also too fragile. This is also the case with a 0.4 nozzle, where aberrations remain. Ideally, tree supports should also have stronger double walls for better reliability. At this point, I have to wait for supports that are truly dependable.

I must admit I was surprised to read that this is considered an old bug. I would be more understanding if users could freely choose other slicers, but since Bambu Studio is the required slicer for these printers, it is difficult to accept that such a long-standing issue has not been resolved yet.

Supports are a fundamental feature of any slicer, and I believe fixing them should be a priority before adding new functionalities. I truly hope this will be addressed soon, as it would greatly improve reliability and user confidence.

Thank you for your work and for continuing to improve the software.