bambulab/BambuStudio #8098: global print presents not syncing

Hofmangearsupply, 2025-09-04:
### Bambu Studio Version

2.2.1.60

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 15.6.1

### Additional system information

Mac mini M4
some of my presents for the global Process settings are there but the main one I use for 90% of my prints is not showing up for my P1S/X1 series printers. I save it , restart the app and it's gone - I can see it on the Batch Presents Management menu , I have found it in the configuration folder. It shows up on my A1 printers and my H2S printer but is now disappearing on my P1S/X1 series printers - I think there is a new conflict with the H2S printer just added - this is a brand new problem I have encountered. How do I fix it?

### Printer

P1S/X1C 

### How to reproduce

1. click drop down menu Process (Global) to select print user presents - only some of my presents are showing. Not all of them. 

### Actual results

idk. 

### Expected results

bring back my user presents.

### Project file & Debug log uploads

[Untitled.3mf.zip](https://github.com/user-attachments/files/22130429/Untitled.3mf.zip)
[log.zip](https://github.com/user-attachments/files/22130430/log.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

DanBao-Bambu, 2025-09-05:
The process preset is associated with the current printer, and the process preset you see is compatible with the current printer. The process user preset saved under the current printer is compatible with the current printer, and switching to another printer may disappear. Was the user preset you saved modified and saved during P1S/X1?

Hofmangearsupply, 2025-09-05:
Hello. Yes it was saved and modified with the p1/x1 printers. I’ve been using this present and variants since the kickstarter. I have a H2S on order and have installed the associated profile. I then saved a process using the H2S and named it the same title. I think that is the issue. The software has since overwritten my original profile and connected it to the H2S printer. This process shows when I select the H2S but disappears when I change to my X1C or P1S.

DanBao-Bambu, 2025-09-09:
Using the same name can cause problems, as the official Process Presets all have an @ model suffix to distinguish between different Process Presets. Using the same name for Process Presets of two incompatible models will result in the previous one being replaced, so the Process Preset of X1C is missing. Suggest using the @ model suffix to distinguish Process Presets for different printers

Hofmangearsupply, 2025-09-09:
Yes, thank you. I figured out this was the issue. Gerrit Hofman