bambulab/BambuStudio #8100: AMS Sync not showing correctly

NVNDO, 2025-09-04:
### Bambu Studio Version

2.2.1.60

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 10

### Printer

P1S

### How to reproduce

Sync AMS of different printers, most of the time they don't get displayed correctly. 

### Actual results

I have two P1S loaded with different amount of filament and types.
(P1S ONE) is loaded only with TPU (slot 2)
(P1S TWO) is loaded with TPU (slot 1+2) and PETG (slot 3+4)

I prepared and started a print with my P1S ONE only printing TPU - worked like expected.

Then, when preparing the second print on my P1S TWO, - when going to " Sync info" no filaments of the connected AMS showed (although they did show in the "Device-tab") Instead it showed only the TPU of my other printers AMS. 

If one ignores the fact that the shown filaments don't correspond to that printer and just click "Overwriting" it does sync correctly with the available colors in that particular AMS - But still, **it shows incorrectly.** 

See Video: https://youtu.be/WnEYSNAfPl0

### Expected results

"Sync info" should display the available AMS colors correctly. 

### Project file & Debug log uploads

[Log AMS.zip](https://github.com/user-attachments/files/22136310/Log.AMS.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

NVNDO, 2025-09-05:
@rouyuanzi-bambu 

I know all that, but thank you anyways for confirming, 

My "Project-file" is no Project file. I don't even have ABS, nor have I ever printed with ABS. 

The file, like I said at the end - comes directly from Fusion 360. 
I don't think Fusion adds any filament- details to an .STL

Maybe Bambu Studio just invents itself types of filament then there aren't any... I don't know - but ABS show up a lot - and like I said, I have never ever printed with AMS and therefore have never saved file with that filament. 

What about the first section of my thread? The "AMS sync" problem.

Haidiye00, 2025-09-09:
Hello, let's analyze