bambulab/BambuStudio #8107: Bed Temp Resetting

SD-Bonz, 2025-09-04:
### Bambu Studio Version

2.2.1.60

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

After I open studio and complete one print, I can no longer set the bed temperature to anything - when I enter a number it will reset to 0.
This is fixed by closing and reopening studio, but as soon as I complete one print it will have the issue again.

### Printer

Bambu Labs P1S

### How to reproduce

1. Complete one print
2. Try to set print bed temperature

### Actual results

The print bed temperature you just set will go back to 0.

https://github.com/user-attachments/assets/03b359ec-0324-45d4-bf81-8c2b69c62735

### Expected results

The print bed temperature should then adjust to the new set temperature.

### Project file & Debug log uploads

[SM5 v3.zip](https://github.com/user-attachments/files/22145686/SM5.v3.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

rouyuanzi-bambu, 2025-09-05:
Hello, thank you for your feedback. We were unable to reproduce the issue you described in version 2.2.1.60. To help us with further analysis, could you please provide the log file ?
The steps to obtain the logs are as follows：https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log

siegmour, 2025-09-09:
> Hello, thank you for your feedback. We were unable to reproduce the issue you described in version 2.2.1.60. To help us with further analysis, could you please provide the log file ? The steps to obtain the logs are as follows：https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log

I can confirm a simialr issue on my A1 Mini. When I change the temp via Bambu Studio, it will revert to the previously set temperature in about a second or so, instead of applying the new temperature.

Restarting Bambu Studio doesn't fix it for me.

Attaching the log file.

[debug_Wed_Sep_03_09_34_49_50708.log.zip](https://github.com/user-attachments/files/22238228/debug_Wed_Sep_03_09_34_49_50708.log.zip)

zwamdurkel, 2025-09-19:
Just wanted to chime in. Same issue for me. Latest firmware of the P1S and BambuStudio. When changing the bed temperature, it just goes back to whatever it was. Setting the bed temperature using the Handy app works fine, but it's a hassle. I've been having this issue for multiple releases of BambuStudio. Not sure when it started

rouyuanzi-bambu, 2025-09-22:
Thank you for your feedback. We have forwarded the issue to the relevant development team, and it is currently under investigation and resolution. We appreciate your patience.

Skye-Bambulab, 2026-11-16:
Hello,we'll close this issue for now.

Regarding this feature request, our final assessment is as follows:
1.Linux-based models: Native support is currently available.

2. New RTOS-based models: We are actively evaluating the possibility of adding support for these devices.

3.A1/P1 Series: Unfortunately, due to significant hardware resourceand firmware code size limitations, we cannot implement this feature on these legacy models. They will remain unsupported for this specific function.

If you have other questions, feel free to reopen it or create a new issue. We'll behappy to continue investigating.