bambulab/BambuStudio #8112: Failed to compile debug version

logic-wei, 2025-09-05:
### Bambu Studio Version

02.02.01.60

### Where is the application from?

Bambu Lab github releases

### OS version

ubuntu 22.04

### Additional system information

ubuntu 22.04

### Printer

p1s

### How to reproduce

 I want to build debug version to enable -g flags. Following the step below:
1. ./BuildLinux.sh -u
2. ./BuildLinux.sh -dsib
For a while, the errors occur.

### Actual results

Failed to compile

### Expected results

compile pass

### Project file & Debug log uploads

CMake Error at src/CMakeLists.txt:113 (add_executable):                                                                                                                                       
  Target "BambuStudio" links to target "Qhull::qhullcpp" but the target was                                                                                                                   
  not found.  Perhaps a find_package() call is missing for an IMPORTED                                                                                                                        
  target, or an ALIAS target is missing?                                                                                                                                                      
                                                                                                                                                                                              
                                                                                                                                                                                              
CMake Error at src/CMakeLists.txt:113 (add_executable):                                                                                                                                       
  Target "BambuStudio" links to target "Qhull::qhullstatic_r" but the target                                                                                                                  
  was not found.  Perhaps a find_package() call is missing for an IMPORTED                     
  target, or an ALIAS target is missing?                                                       
                                                                                               
                                                                                               
CMake Error at src/libnest2d/CMakeLists.txt:24 (add_library):             
  Target "libnest2d" links to target "Qhull::qhullcpp" but the target was not                  
  found.  Perhaps a find_package() call is missing for an IMPORTED target, or                  
  an ALIAS target is missing?                                                                  
                                                                                               
                                                                                               
CMake Error at src/libnest2d/CMakeLists.txt:24 (add_library):         
  Target "libnest2d" links to target "Qhull::qhullstatic_r" but the target                     
  was not found.  Perhaps a find_package() call is missing for an IMPORTED                     
  target, or an ALIAS target is missing?                                                       
                                                                                               
                                                                                               
CMake Error at src/libslic3r/CMakeLists.txt:480 (add_library):          
  Target "libslic3r" links to target "Qhull::qhullcpp" but the target was not
  found.  Perhaps a find_package() call is missing for an IMPORTED target, or                  
  an ALIAS target is missing?                                                                  
                                               

CMake Error at src/libslic3r/CMakeLists.txt:480 (add_library):
  Target "libslic3r" links to target "Qhull::qhullstatic_r" but the target
  was not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

CMake Error at src/slic3r/CMakeLists.txt:642 (add_library):
  Target "libslic3r_gui" links to target "Qhull::qhullcpp" but the target was
  not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

### Checklist of files to include

- [ ] Log file
- [ ] Project file

tqtb89, 2025-09-26:
> solved
> 
> ```
> # reboot
> rm -rf ./deps/build
> rm -rf ./build
> sudo ./BuildLinux.sh -ur
> ./BuildLinux.sh -d
> ./BuildLinux.sh -bs
> ```
> 
> But, crashed when slicing any model at:
> 
> [BambuStudio/src/libslic3r/EdgeGrid.cpp](https://github.com/bambulab/BambuStudio/blob/4f0f0f3e97400629205048995c77b5eedd2768e5/src/libslic3r/EdgeGrid.cpp#L68)
> 
> Line 68 in [4f0f0f3](/bambulab/BambuStudio/commit/4f0f0f3e97400629205048995c77b5eedd2768e5)
>  assert(*begin != end[-1]);

@MackBambu Did I miss something?

lanewei120, 2025-09-28:
There are lots of assertions that should be skipped, 
You can just comment out these asserts if using debug building