bambulab/BambuStudio #8126: 2.2.1.60 crashing HARD and powers off when I hit slice

ShortyCM, 2025-09-08:
### Bambu Studio Version

2.2.1.60

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 24H2

### Additional system information

12900KS
64 GB DDR5
3080 Ti

### Printer

X1C

### How to reproduce

All I do is hit slice.  The instant I do that the machine powers off and stops repsonding to the power button. This has been happening randomly ever since the release of 2.2.1.60 and never happened before that.

### Actual results

Machine crashes HARD, instantly the power completely shuts off.  Won't respond to the power button.  I have to disconnect the power cable for a few seconds to get it to respond to the power button again so that the machine turns back on.

### Expected results

Slicing without crashing the machine into an unrecoverable power state.

### Project file & Debug log uploads

[projects.zip](https://github.com/user-attachments/files/22200145/projects.zip)

[logs.zip](https://github.com/user-attachments/files/22200151/logs.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

ShortyCM, 2025-09-15:
> Additionally, providing dump information will help pinpoint the problem. https://wiki.bambulab.com/en/bambu-studio/troubleshoot/crash-freeze-issue If you can't find the dump file in the file path listed in the wiki, it may indicate a more fundamental driver, hardware, or operating system issue. Try running other CPU or hardware testing programs. Intel provides a processor stability checker that can verify the stability of your processor: https://www.intel.com/content/www/us/en/download/15951/intel-processor-diagnostic-tool.html Of course, you can also try other CPU stress testing tools, such as Prime95. (We recommend using multiple methods and performing more extensive and longer tests to fully verify stability.)
> 
> Thank you for your efforts

There's nothing wrong with my machine.  And there are no dump files.  As I stated, the machine instantly shuts off the power when the issue occurs.  You can't write a dump file with no power. I believe I included everything that was available.  Whatever it is doing is pretty harsh since, as I said, the machine won't even power back on when it does this.  The power button no longer functions until I unplug the power cord.  The machine appears completely dead until doing so.  If you guys figure there's something wrong with how you're using the video card then maybe your installer should be the one that is ensuring the Nvidia profile for your executable is set to not use the default multithreading.  The nature of this power shut off behaviour is putting other data on my machine at risk.  If this video card multithreading is even the problem in the first place.

Diatom-Banbu, 2025-09-16:
> > Additionally, providing dump information will help pinpoint the problem. https://wiki.bambulab.com/en/bambu-studio/troubleshoot/crash-freeze-issue If you can't find the dump file in the file path listed in the wiki, it may indicate a more fundamental driver, hardware, or operating system issue. Try running other CPU or hardware testing programs. Intel provides a processor stability checker that can verify the stability of your processor: https://www.intel.com/content/www/us/en/download/15951/intel-processor-diagnostic-tool.html Of course, you can also try other CPU stress testing tools, such as Prime95. (We recommend using multiple methods and performing more extensive and longer tests to fully verify stability.)
> > Thank you for your efforts
> 
> There's nothing wrong with my machine. It is not overclocked. It is not dealing with insufficient cooling or any other nonsense. It is a perfectly functioning machine other than when encountering this issue. And there are no dump files. As I stated, the machine instantly shuts off the power when the issue occurs. You can't write a dump file with no power. I believe I included everything that was available. Whatever it is doing is pretty harsh since, as I said, the machine won't even power back on when it does this. The power button no longer functions until I unplug the power cord. The machine appears completely dead until doing so. If you guys figure there's something wrong with how you're using the video card then maybe your installer should be the one that is ensuring the Nvidia profile for your executable is set to not use the default multithreading. The nature of this power shut off behaviour is putting other data on my machine at risk. If this video card multithreading is even the problem in the first place.

We sympathize with your situation and recommend backing up your computer in advance to avoid data loss. However, regarding the issue you described, we were unable to reproduce the crash using the same version, indicating that the problem lies with your computer. We offer the following explanations:
1. Regarding multithreading issues, multithreading is never the problem itself. We recommend that users experiencing crashes adjust multithreading settings to avoid the already unstable processor "core" (Some Intel CPUs have unstable issues). However, disabling multithreading for users with faulty computers would increase slicing times for everyone, which is not worth the cost.
2. I understand that a direct power outage may not leave any memory dump files, but this does make troubleshooting unreproducible issues difficult. This issue cannot be reproduced in our testing, so we rely on you to provide more detailed information.
3. The power outage issue you describe may be related to the hardware, drivers, operating system, or program installation. I once encountered a computer with an underpowered power supply. The power circuit was aging, but it functioned normally until I ran Furmark to test it. The increased power consumption caused the power supply voltage to become unstable, resulting in a power outage. Therefore, we recommend trying some load tests, such as Prime95, Furmark, and CPU-z, to stress your CPU or graphics card and verify adequate power supply and hardware stability.
4. Additionally, corrupted system files or corrupted software installations can cause unexpected issues. For software, you can uninstall Studio, re-download the full installation package, and reinstall to repair any corruption. For Windows systems, you can review Microsoft's tutorial on checking system integrity (file system integrity and system file integrity):
https://learn.microsoft.com/en-us/windows-server/administration/windows-commands/chkdsk?tabs=hdd%2Cevent-viewer
https://support.microsoft.com/en-us/topic/use-the-system-file-checker-tool-to-repair-missing-or-corrupted-system-files-79aa86cb-ca52-166a-92a3-966e85d4094e

Diatom-Banbu, 2025-09-16:
> 
> THERE. IS. NOTHING. WRONG. WITH. MY. COMPUTER.

Now that you have confirmed that your computer is normal, you can try reinstalling Studio. 
In addition, does your problem occur every time you slice, or does it only occur in random situations?

ShortyCM, 2025-09-16:
> Now that you have confirmed that your computer is normal, you can try reinstalling Studio. Errors in network transmission may damage the integrity of the installation package. Re-downloading may help. In addition, does your problem occur every time you slice, or does it only occur in random situations?

Compressed files have a hash attached to them to ensure that decompression was successful.  Silently corrupt installation files are highly unlikely.  And the original report states quite clearly that this is a random thing.  Whenever it does actually happen, the outcome is the same.  Instantaneous power off with a machine that is no longer responding to the power button, requiring the disconnection of the power cord from the wall outlet to correct.