bambulab/BambuStudio #8139: Printing Defect When Using the Fuzzy Skin Painting Tool

kle7rx, 2025-09-09:
### Bambu Studio Version

2.2.1.60

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

Intel(R) Core(TM) i7-4790K CPU @ 4.00GHz, 32Gb, Nvidia GeForce GTX 750

### Printer

Bambu Lab X1C

### How to reproduce

1. In the print profile settings, go to the Others tab and enable the Fuzzy Skin option by setting it to Contour.
2. Set the parameters: Fuzzy skin point distance = 0.1 mm and Fuzzy skin thickness = 0.2 mm.
3. Then, switch the Fuzzy Skin option back to None.
4. Select the model.
5. Choose the Fuzzy Skin Painting Tool and use the Fill function to apply fuzzy skin to a surface of the model.
6. Slice the model and send it for printing.

### Actual results

On a circular part with a diameter of 161 mm, the defect is approximately 220 mm long and appears as if the filament in this area has Fuzzy Skin point distance and Fuzzy Skin thickness settings that differ from those specified in the print profile. When re-slicing and printing the same model, the defect appears in different locations, sometimes after several hours of printing.

IMPORTANT NOTE: If the global Fuzzy Skin option in the Others tab is left enabled and the Fuzzy Skin Painting Tool is NOT used, the defect does not occur.

Actions taken before reporting the issue: 
1. Performed hotend and extruder cleaning.  
2. Used filaments from different manufacturers.  
3. Printed from different AMS slots.

[image]
[image]
[image]
[image]

### Expected results

The printed model should have a consistent fuzzy texture on the painted areas, matching the parameters set in the painting tool (or inheriting the last set parameters of 0.1mm/0.2mm correctly).

### Project file & Debug log uploads

[Fuzzy Skin Painting Tool.zip](https://github.com/user-attachments/files/22242952/Fuzzy.Skin.Painting.Tool.zip)
[log.zip](https://github.com/user-attachments/files/22242951/log.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

JiaYun-Bambu, 2025-09-16:
I used PETG material to print this model and haven't encountered the same issue so far. 
[image]
You can send us the GCode file to help us further analyze the possible causes.
[image]

kle7rx, 2025-09-16:
Yes, I managed to print this small part several times without defects. I am attaching a larger part. Using the Fuzzy Skin Painting Tool, I was unable to print this part well.

[Fuzzy Skin Painting Tool.gcode.zip](https://github.com/user-attachments/files/22356655/Fuzzy.Skin.Painting.Tool.gcode.zip)

kle7rx, 2025-09-16:
Yes, I managed to print this small part several times without defects. I am attaching a larger part. Using the Fuzzy Skin Painting Tool, I was unable to print this part well.

kle7rx, 2025-09-16:
...and thank you very much for helping me with my problem)

kle7rx, 2025-09-16:
I am attaching the project files. I was able to print the models with a small Fuzzy Skin Painting Tool fill area several times without a defect. On two rings from the project with a large fill area, the defect appears every time.

[Arabesque Lamp.zip](https://github.com/user-attachments/files/22369365/Arabesque.Lamp.zip)

[ArbsqLmp_Base Lid 72.gcode.zip](https://github.com/user-attachments/files/22369391/ArbsqLmp_Base.Lid.72.gcode.zip)

JiaYun-Bambu, 2025-09-17:
> Here is a photo of the model where the defect appeared on the very top surface layer with the Fuzzy Skin Painting Tool fill applied.
> 
> [image]
> 
> This is the same model. A photo of the back side where the defect is not present.
> 
> [image]

Which location on which plate of the model is this?

kle7rx, 2025-09-26:
I suspect the Arachne engine doesn't play well with the Fuzzy Skin Painting Tool. However, I can't say for sure yet, as I haven't had time for large-scale testing.