bambulab/BambuStudio #8174: Bambu Studio Crashed at end of Print Job

rdforbes, 2025-09-14:
### Bambu Studio Version

2.2.1.60

### Where is the application from?

Bambu Lab Official website

### OS version

macOS Sequoia 15.6.1

### Additional system information

MacStudio 2023 32GB Memory

### Printer

Bambu Lab A1 w/ AMS Lite

### How to reproduce

1. downloaded a file from MakerWorld
2. printed it successfully
3. Bambu Studio crashed at the end of the print job

### Actual results

Bambu Studio crashed; macOS displayed a Crash Report

### Expected results

Bambu Studio should not crash.

### Project file & Debug log uploads

[Archive.zip](https://github.com/user-attachments/files/22320132/Archive.zip)

[Crash Report.rtf](https://github.com/user-attachments/files/22320135/Crash.Report.rtf)

[A1toolBOX.3mf.zip](https://github.com/user-attachments/files/22320137/A1toolBOX.3mf.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

Diatom-Banbu, 2025-09-26:
Hello, thank you for your feedback. We haven't been able to reproduce this issue on macOS and may need more information to troubleshoot the issue.
1. Do you frequently experience this crash? How often does it occur?
2. Have you tried uninstalling and installing a new version of the software?
3. Have you tried deleting and reinstalling the network plugin?

rdforbes, 2025-09-27:
> Diatom-Banbu
>  left a comment
> (bambulab/BambuStudio#8174)

1. I am experiencing a crash intermittently, about once every ten times I use my Bambu Studio. 
2. The random crashes started when I updated to version 2.2.2.56.
3. Please send me instructions on how to delete and reinstall the network plugin.

Thank you for the support,
Ron Forbes

========================

Diatom-Banbu, 2025-09-28:
On macOS, we recommend installing the latest version of the software directly. The network plugin will also be updated with the latest version.
Uninstalling and reinstalling the latest version may help. The next release will fix the crash caused by this issue.
(If your situation is due to corrupted installation files, re-downloading the current version may also fix the issue. However, due to insufficient information, we cannot confirm whether this will work. We will add protection to prevent crashes in the next release.)