bambulab/BambuStudio #8175: Unload filament after stopping a print.

NVNDO, 2025-09-14:
**My feature request / problem:**
Its super frustrating that every time I stop a print, which mostly happens right before starting a print, that the filament previously loaded stays inside the printer. When stopping it should get recovered.
There is also no unload button / option. 

When after stopping a print, I start a new print - This is not a problem, as the previously loaded filament get unloaded and the new one loads, BUT if I want to change the spool of that previously stopped print, I can't. 

I do understand this logic (filament stays) when "pausing".
But when I "stop" a print, I want to start over, hence (filament should get unloaded). 

**Describe the solution you'd like**
Unload filament from printer (back to AMS) after stopping a print. 

**Describe alternatives you've considered**
None

**Additional context**
It's not a bug, but far from logic and should get some attention.

NVNDO, 2025-11-17:
Hey @JinGangBambu & @JohnSmithCYM 

Any news on this one? Will this be implemented or given thoughts to? 
Would be great. 

Thank you.

JohnSmithCYM, 2025-11-18:
Thank you for your feedback. The relevant request has been forwarded to the firmware team. Please wait for their follow-up response.