bambulab/BambuStudio #8180: High Flow settings keep resetting in material profiles

fullHundo, 2025-09-15:
### Bambu Studio Version

2.2.1.60

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab H2D

### How to reproduce

1. Verify high flow hotends installed and detected
2. Click to edit filament preset
3. Click drop-down menu to swap from "Direct Drive Standard" to "Direct Drive High Flow"
4. Increase Max Volumetric Speed value
5. Save profile
6. Close application then reopen
7. Recheck that value has not reverted to default value
8. Repeat this multiple times. It usually saves the value just fine for at least the rest of the day, but almost always has been reset at some point within a day or a few days

### Actual results

The adjusted max volumetric speed value for high flow hotends keeps resetting itself eventually. It generally doesn't happen right away, I can close and reopen the application on multiple computers and it still retains the updated value, but eventually, I will check and the value has reverted back to default. I haven't tested to see how quickly, but it always happens eventually within a few days at most. This results in me constantly having to check that my adjusted value is still saved before every print with a high flow hotend.

### Expected results

Expected result is that once a new max volumetric speed value is saved for the high flow hotend portion of a filament profile, it will just be saved permanently and not need to be rechecked constantly before printing.

### Project file & Debug log uploads

not needed

### Checklist of files to include

- [ ] Log file
- [ ] Project file

0IlIlIlIlIlll0, 2025-11-03:
Seeing the same behavior for both process settings and filament settings.  When flow is set to "high flow" and saving attempting to save a user preset, bs acts like it has saved the settings, but after closing bs, and reopening, the settings are reverted back to the previous settings.

st3venb, 2026-11-04:
This bug is still present in 2.9.1.70.