bambulab/BambuStudio #8194: Error code 0300-3100-0001-0001 after latest update

FvdM-GH, 2025-09-16:
### Bambu Studio Version

2.2.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

W11 64-bit

### Additional system information

Processor	Intel(R) Core(TM) i7-9700 CPU @ 3.00GHz (3.00 GHz)
Installed RAM	48.0 GB (47.8 GB usable)
System type	64-bit operating system, x64-based processor

### Printer

Bambu Lab P1S with AMS-1

### How to reproduce

Start 3D print, error pops up.

### Actual results

Normal start of the 3D Print, but error keeps popping up in the display screen and in the Assistant (HMS).

### Expected results

No errors.

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/22359166/log.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

FvdM-GH, 2025-09-16:
Error also pops up in the Bambu Handy app.

JinGangBambu, 2025-09-16:
This looks like it might be a printer-related issue.
We have a specific process for handling these. Please follow the instructions below to create a post-sales order. The firmware team will resolve the problem based on the post-sales order and they does not track orders on GitHub.

Please upload log through Handy APP, and give us your ticket number

Upload log files through Bambu Handy
Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
To upload files, go to Me → Support Tickets → Choose the ticket for log upload and proceed with the operation.
app_support_ticket

[image]

FvdM-GH, 2025-09-16:
Fixed after checking the print head, the fan connector had popped out of the PWB connctor.
REMARK: The connectors are interconnected very loosely, it almost falls out itself. Slight vibrations might be enough to get this problem.