bambulab/BambuStudio #8205: Non-Bambu filament persists in AMS even after replacing with Bambu filament

bandesz, 2025-09-17:
### Bambu Studio Version

2.2.1.60

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 15.6.1

### Additional system information

-

### Printer

P1S + AMS 1

### How to reproduce

1. Put a non-Bambu filament into the AMS
2. Set the non-Bambu filament profile for the AMS slot in Bambu Studio
3. Replace the non-Bambu filament with a Bambu filament in the AMS

### Actual results

The non-Bambu filament profile is not replaced automatically with a recognised Bambu filament profile. I'm not 100% sure but it seems the AMS does not even try to detect whether it's a Bambu filament?

### Expected results

The Bambu filament profile should be automatically set.

### Project file & Debug log uploads

Issue is not specific to project.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

bandesz, 2025-09-17:
I have to say I'm not 100% sure about the reproduction, but I know it happens regularly and it also happens sometimes with my A1 + AMS Lite combos.

A1: 01.06.00.00, AMS Lite: 00.00.08.08
P1S: 01.09.00.00, AMS-1: 00.01.06.62

JinGangBambu, 2025-09-17:
This looks like it might be a printer-related issue.
We have a specific process for handling these. Please follow the instructions below to create a post-sales order. The firmware team will resolve the problem based on the post-sales order and they does not track orders on GitHub.

Please upload log through Handy APP, and give us your ticket number

Upload log files through Bambu Handy
Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
To upload files, go to Me → Support Tickets → Choose the ticket for log upload and proceed with the operation.
app_support_ticket

[image]