bambulab/BambuStudio #8211: Wrong AMS colours when sync

geleto001, 2025-09-17:
### Bambu Studio Version

2.2.1.60

### Where is the application from?

Bambu Lab Official website

### OS version

win11

### Printer

P1S

### How to reproduce

When i sync the colours it shows always the same colours , no matter what I have in the AMS 

### Actual results

Completey wrong colours .  Not a single match

### Expected results

The right colours obviously 

### Project file & Debug log uploads

it is not a project it does it every time 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

JinGangBambu, 2025-09-18:
Hi, thank you for reporting this issue!
To help us better understand and reproduce the problem on our side, could you please:

1. Record a short screen video of any steps you take. You can use a simple screen recorder such as [ScreenToGif](https://github.com/NickeManarin/ScreenToGif/releases/tag/2.41.1).
2. Provide the log: [Export Bambu PC software log](https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log)

This information will help us locate the cause more quickly and provide an effective solution. Thank you so much for your cooperation!

Haidiye00, 2025-09-19:
ok

Haidiye00, 2025-11-19:
https://github.com/bambulab/BambuStudio/releases/tag/v02.04.00.70
Hello, the new version has been released, please update.
If there are any further questions, let's continue the discussion. Thank you