bambulab/BambuStudio #8224: Text is no longer centered on objects

mark6B, 2025-09-18:
### Bambu Studio Version

2.2.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

macOS Sequoia 15.5

### Additional system information

Hardware Overview:

  Model Name:	Mac mini
  Model Identifier:	Macmini9,1
  Model Number:	MGNT3LL/A
  Chip:	Apple M1
  Total Number of Cores:	8 (4 performance and 4 efficiency)
  Memory:	8 GB
  System Firmware Version:	11881.121.1
  OS Loader Version:	11881.121.1

Apple M1:

  Chipset Model:	Apple M1
  Type:	GPU
  Bus:	Built-In
  Total Number of Cores:	8
  Vendor:	Apple (0x106b)
  Metal Support:	Metal 3

### Printer

Bambu P1S

### How to reproduce

Add an object
Open the Text Shape tool
Enter text

In previous versions, the text could be centered on the object by setting the "X" or "Y" position (based on orientation) to 0.00.
With this version, setting Y to 0.00 does not center the text in the object.

### Actual results

Depending on the text length, I have to manually adjust the text to center it in the object.  I can use the "Center" capability on the object itself, but that seems like an unnecessary step since it always worked correctly in prior version.

### Expected results

Text objects should be centered when their X & Y coordinates are 0.0, 0.0.

### Project file & Debug log uploads

[Bug Report 2.2.2.56.3mf.zip](https://github.com/user-attachments/files/22414483/Bug.Report.2.2.2.56.3mf.zip)

[debug_network_Thu_Sep_18_15_45_33.log.enc.cpgz.zip](https://github.com/user-attachments/files/22414506/debug_network_Thu_Sep_18_15_45_33.log.enc.cpgz.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

JinGangBambu, 2025-09-19:
Hello,
Thanks for your feedback. To help us better understand and reproduce the problem on our side, could you please:
Record a short screen video of any steps you take. You can use a simple screen recorder such as [ScreenToGif](https://github.com/NickeManarin/ScreenToGif/releases/tag/2.41.1).
This information will help us locate the cause more quickly and provide an effective solution. Thank you so much for your cooperation!

Haidiye00, 2025-09-19:
Hello, we will conduct an analysis

SirNahka, 2025-09-29:
I have the same issue, text_shape object is set to X=0.00 and Y=0.00, but text is not centered on main object.

[image]

After right clicking the text_shape object in the Process listing and using Center in the context menu, these are apparently the "correct" centered coordinates:

[image]

Haidiye00, 2025-10-15:
https://github.com/bambulab/BambuStudio/releases/tag/v02.03.00.70
Hello, the new version has been released. Please update the software. Thank you~

closed 2025-10-15 by Haidiye00