bambulab/BambuStudio #8228: 切片或下发打印时闪退

GarryLLL, 2025-09-19:
### Bambu Studio Version

2.2.2.56

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11 家庭中文版24H2

### Additional system information

CPU:Intel(R) Core(TM) i7-14700 (2.10 GHz)
RAM:32.0 GB

### Printer

P1S

### How to reproduce

打开“拉布布”3mf文件，点击切片，切片完成时点击打印

### Actual results

切片完成时闪退或者切片完成后点击打印时闪退，桌面端几乎毕现。

### Expected results

期望能正常下发打印

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/22418204/log.zip)

[2-眼睛&嘴.zip](https://github.com/user-attachments/files/22418207/2-.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

troublerandomguy31, 2025-09-19:
Maybe this update helps.
[3d0d8bcb2baa.zip](https://github.com/user-attachments/files/22418222/3d0d8bcb2baa.zip)

Diatom-Banbu, 2025-10-24:
您好，感谢您的反馈。经过排查，您的崩溃发生的位置似乎并不固定，有些甚至在ntdll.dll中，难以揭示软件本身存在的问题，同时，我们排查中并没有复现您所说的崩溃情况，能够正常完成切片并发送，因此可能需要您提供更多信息或尝试，以排查问题所在。
1. 您是否确定是在切片“眼睛&嘴”的时候发生问题？似乎并没有提供前面提到的“拉布布”3mf文件
2. 下载安装中可能存在传输错误，尝试卸载并重新下载安装软件，可能解决崩溃问题
3. 系统文件损坏也可能造成软件不稳定，尝试使用来自windows系统的 系统文件完整性检查（System File Integrity Check，命令：sfc /scannow）以及文件系统完整性检查（File System Integrity Check，命令：chkdsk C:，或其他磁盘），以确定您的系统环境正常
4. 如果您使用nvidia显卡或者intel的13、14代CPU（通常出现在带有K的型号上），可能存在某些硬件稳定性问题，查看该指南中的信息并尝试排查可能有助于您确定问题所在 ：https://wiki.bambulab.com/zh/bambu-studio/troubleshoot/win-crash-when-slicing