bambulab/BambuStudio #8238: Adopt "Consistent Surface" and "Short Specific Travel Acceleration" from PrusaSlicer

lorenzofanchi, 2025-09-20:
PrusaSlicer recently introduced two features that aim to significantly improve the quality of printed models: Consistent Surface and Short Specific Travel Acceleration. The former reduces/solves one of the main issues with inconsistent gloss and matte surfaces on the same model.

The latter should also drastically improve outer surface quality, as it reduces ringing on outer walls.

Top - OrcaSlicer/Bambu Studio default profiles.
Bottom - the new cooling logic and the short specific travel acceleration in PrusaSlicer 2.9.3.

[image]

Read more about both [here](https://blog.prusa3d.com/new-in-prusaslicer-consistent-surface-finish-and-nerfing-vfas_120400/)

lorenzofanchi, 2025-11-26:
I was a bit disappointed with VFA's after receiving my H2D, so I hope Bambu can optimize Studio to minimize them 🙏🏼

lorenzofanchi, 2025-11-26:
Very curious! Would be awesome to see a side by side comparison of the same printed model; one with the current version of Studio and one with your changes!

raulp, 2025-11-27:
@lorenzofanchi thanks! Let me check, i might have a problem with LAN mode https://github.com/bambulab/BambuStudio/issues/8887. Let me sort that one first then i can try. Appreciate you sending the model.

raulp, 2025-11-29:
unfortunately, i tried everything i could to make this work, but i'll have to ask for some help. I fixed the bug regarding LAN mode. I added my printer, but, because it's a local build, i can't send it to print, LAN mode or not. All i get is the fact that the app is unsigned and can't print. I tried INTERNAL build, same error.

[image]

Not sure if anyone else has any feedback about this, but i'll keep trying later.

lorenzofanchi, 2025-11-30:
I recall there is a setting called "Develop mode" in Bambu Studio Settings; maybe try enabling that?

[image]

raulp, 2025-12-05:
**Slicer settings** (make sure development mode is on):

[image]

[image]

Here's some **slicer preview** pictures with the features **ON**:

[image]

[image]

And here's with the features **OFF**:

[image]

[image]

Slicer clearly shows it better than i can explain, but what's going on is that without the new option, everything is slowed down to the same speed. So walls, infill, etc is slowed down to match the layer time. With the option turned on, outer walls speed remains consistent with the settings, and everything else is slowed down.

I need to better test the short acceleration feature, as this test rectangle is not enough. 

For anyone else that wants to try this, i used the workflows from Bambu to create builds. They can be downloaded here, by scrolling down to Artifacts:
- macos/windows: https://github.com/raulp/BambuStudio/actions/runs/19803533385
- linux: https://github.com/raulp/BambuStudio/actions/runs/19803533290 (it's queued, not sure when the build will be ready)

I will post some more pictures later this weekend, because the test that i did showed mostly VFA due to lower speeds, so i want to play more with the cube until i don't see VFA on either print

[image]

lorenzofanchi, 2025-12-05:
This is great! Looks like a massive difference in quality. I'm assuming top is with the option off and bottom is with the option on? How much is the difference in printing time between the two?

I'll definitely give this a spin soon. Thank you for putting in the time to make the build. If you have more time, you should def tag a dev and/or open a PR 🥇

lorenzofanchi, 2025-12-05:
I can see:
- BambuStudio_Mac_arm64_PR-1
- BambuStudio_dep_mac_arm64_20251130

Are they equivalent? The former is much bigger (1.36 GB)

lorenzofanchi, 2025-12-05:
Re testing the effect of Short Specific Travel Acceleration on VFA's, I see what you mean: since the Consistent Surface option is effectively making the outer wall speed much slower, it's not a fair comparison.. I'll think about how to test this properly.

raulp, 2025-12-06:
I think we need the community to test this out a bit more. More than happy to open the PR though, i finished a long 14hrs print and  used both settings + Smooth speed discontinuity and it worked fine. Extra eyes won't hurt though. 

I still want to figure out a test for the short acceleration. Did you came up with something for that?

lorenzofanchi, 2025-12-06:
> I still want to figure out a test for the short acceleration. Did you came up with something for that?

I think one way to do it would be to:
- disable **Consistent Surface**
- print one block with **Short Specific Travel Acceleration** set to `250`
- print one block with **Short Specific Travel Acceleration** set to `10000` (i.e. the same as the normal **Travel Acceleration** on my H2D)

Thoughts?

P.S. I'll start testing next week, as this weekend I won't be able to find time.

lorenzofanchi, 2025-12-06:
Unfortunately I can't get the build to work. I keep repeatedly getting this modal, even after clicking Allow:

[image]

I gave the app Full Disk Access, but nothing changes.

raulp, 2025-12-06:
Hey, i am not sure why, but i get it as well, i just click allow. That's just reading your recent project files, one by one. Not exactly sure why it does that, but just clicking Allow worked. Hope you don't have 100 projects haha.

Your screenshot makes sense, i'll also try to look up some models and see if i can spot differences like that.