bambulab/BambuStudio #8241: Bambu Sutdio - waits on Loading Configuration a long time home page does not load so cannot log in!

theboyknowsclass, 2025-09-20:
### Bambu Studio Version

2.2.2

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 - latest

### Additional system information

CPU: AMD 9950X3D
GPU RTX 5090
RAM 96GB

### Printer

Bambulab X1C

### How to reproduce

Load Bambu Studio.
Wait (a long time - couple of minutes)

### Actual results

Result:
The Shell loads... all tabs work *EXCEPT* the home tab which is totally blank

### Expected results

Expected Result:
Shell loads normally with a populated homepage

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/22439525/log.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

theboyknowsclass, 2025-09-20:
btw if it's GPU relsted 

Nvidia Studio Driver - 581.29

Mujin-code, 2025-09-22:
Hello, thank you for your feedback. You can follow this process to check if the problem can be solved. If the problem still exists, please feel free to provide further feedback：
https://wiki.bambulab.com/zh/bambu-studio/troubleshoot/install-issue

theboyknowsclass, 2025-09-22:
this is a new install of windows and bambu studio with no previous install - fresh install

when i try and install the edge runtime i get his message
 pop-up message says “This computer already has a more recent version of Microsoft Edge. If the software is not working, please uninstall Microsoft Edge and try again.”

I sthere a specific version of edge runtime that the app needs?   if so it might be worth using the fixed installer - i think you have the evergreen one?

[image]