bambulab/BambuStudio #8251: Camera feed turns off while printing when there is no other activity

gloriouslyawkwardlife, 2025-09-21:
### Bambu Studio Version

2.2.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS Tahoe 26.0 (25A354)

### Additional system information

Macbook Air 13-inch, M3, 2024 (16 GB RAM, 512 GB Storage) 

### Printer

Bambu Lab X1 Carbon

### How to reproduce

- When printing, go to the **Device** screen.
- Click on the green play icon below the preview window to begin viewing camera.
- After a time, the preview shuts off because it believes nothing has been done for a while, but the printer is still printing.

### Actual results

After a time, the preview stops with a message that it has done so because there has been no activity for a time, even though the printer is still printing.  This behavior did not occur in previous versions.

### Expected results

The preview should remain active until the printer has completed the print job.

### Project file & Debug log uploads

This occurs on every project. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Mujin-code, 2025-09-22:
Hi, thank you for reporting this issue!
To help us better understand and reproduce the problem on our side, could you please:

Record a short screen video of any steps you take. You can use a simple screen recorder such as [ScreenToGif](https://github.com/NickeManarin/ScreenToGif/releases/tag/2.41.1).
Provide the studio log: [Export Bambu PC software log](https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log)
This information will help us locate the cause more quickly and provide an effective solution. Thank you so much for your cooperation!

Mujin-code, 2025-09-29:
In the preference settings, there is an option that may affect the use of this feature. You can check if this option is causing the problem. Thank you~

[image]

gloriouslyawkwardlife, 2025-09-29:
Bingo!  Why that's not set on by default…