bambulab/BambuStudio #8259: printables.com files fail to open

tal-p, 2025-09-22:
### Bambu Studio Version

2.2.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

Opening any files from Printables.com using the "Open in Bambu Studio" option results in an error message that the file type is not recognized. However, these are normal stl or 3mf files and should open without issue.

### Printer

Bambu Lab P1S

### How to reproduce

1. Go to any file in Printables.com
2. Go to files tab
3. Select "Open in Bambu Studio"
4. Click "Yes" on popup that file is from unverified source
5. Error dialogue will appear.

### Actual results

The attached error message appears, saying the file format is unknown.

[image]

### Expected results

The file should open normally in Bambu Studio, as they are normal stl and 3mf files. They open in OrcaSlicer using the exact same flow without issue.

### Project file & Debug log uploads

N/A. Files work when downloaded, it's only opening directly from browser that fails.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Mujin-code, 2025-09-22:
Hi, thank you for reporting this issue!
We are unable to reproduce your issue
To help us better understand and reproduce the problem on our side, could you please:

Record a short screen video of any steps you take. You can use a simple screen recorder such as [ScreenToGif](https://github.com/NickeManarin/ScreenToGif/releases/tag/2.41.1).
Provide the studio log: [Export Bambu PC software log](https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log)
This information will help us locate the cause more quickly and provide an effective solution. Thank you so much for your cooperation!

mechaturtles, 2025-10-19:
I have this same issue with downloading STLs from Printables on v2.2.1.60 on Windows 11. I've experienced it for a few months now and the only solution is to download the STL directly and then load into Bambu Studio.

Mujin-code, 2025-12-15:
Hello, we are currently working on resolving this issue. Thank you for your understanding

ScrapyCocos, 2026-01-08:
I just discussed with Prusa customer service about this issue and they said that it’s something in Bambu studio that it’s specifically made for the 3mf’s from prusa not to be compatible. I told them about your finding (@MackBambu ) and they told me to try and load the 3mf manually in Bambu studio and that did not work also. It was throwing the error “3mf file not from Bambu” and the only option given was to load just the geometry.
Looks like the fight between Prusa and Bambu is still there and also looks like the one that keeps the fight going is Bambu Lab unfortunately.

Mujin-code, 2026-01-15:
@mikicvi Hello, can you send us the 3mf that failed to import for us to try?

justincavery, 2026-03-08:
I"m having the same issue, specifically with https://www.printables.com/model/1225275-modular-10-server-rack-mod10/files

DanielCarmingham, 2026-09-04:
> The problem you're seeing is because Printables.com tries to launch a ZIP file, which Bambu Studio does not know how to open. If you look at the request URL, you can see it's trying to launch a file with the wrong protocol, the wrong URL scheme, and the wrong file type extension.
.
.
> (_You could also reasonably argue that this would be REALLY complicated and would break some stuff. e.g. what if one of the 3mf files already has multiple plates? What if you try to 'save' a file that lives in a zip archive? What if Bambu Studio needs to alter the file and commit it to disk, but the file itself is in an ephemeral temp directory and is lost shortly after you close Bambu Studio? Etc._)

@thisnameissoclever You're right, this is a complicated one.  I also have a couple questions about handling zip files:

* Assuming a zip file is extracted before its contents are accessed, should it be extracted to a temp dir or where the .zip file is?
* What happens if there are multiple .3mf files in the zip?  
* Does Prusa Slicer handle .zip files?

It could also be argued that printables.com shouldn't be sending a .zip file to the bambustudioopen:// in the first place, since it's not a Bambu Studio file.

silvetti, 2026-11-09:
I also have the same exact problem.