bambulab/BambuStudio #8290: GLib issue

cinderblockgames, 2025-09-24:
### Bambu Studio Version

2.2.2 Public Release

### Where is the application from?

Bambu Lab Official website

### OS version

LMDE 6 (faye) x86_64

### Additional system information

    .+o++::.      `.:oos+.     OS: LMDE 6 (faye) x86_64 
   :oo:.`             -+oo:    Host: Laptop 16 (AMD Ryzen 7040 Series) AG 
 `+o/`    .::::::-.    .++-`   Kernel: 6.1.0-40-amd64 
`/s/    .yyyyyyyyyyo:   +o-`   Uptime: 23 mins 
`so     .ss       ohyo` :s-:   Packages: 2445 (dpkg), 11 (flatpak) 
`s/     .ss  h  m  myy/ /s``   Shell: bash 5.2.15 
`s:     `oo  s  m  Myy+-o:`    Resolution: 1920x1200 
`oo      :+sdoohyoydyso/.      DE: Cinnamon 6.4.8 
 :o.      .:////////++:        WM: Mutter (Muffin) 
 `/++        -:::::-           WM Theme: Mint-Y-Dark (Mint-Y) 
  `++-                         Theme: Mint-Y-Dark [GTK2/3] 
   `/+-                        Icons: Mint-Y [GTK2/3] 
     .+/.                      Terminal: gnome-terminal 
       .:+-.                   CPU: AMD Ryzen 7 7840HS w/ Radeon 780M Graphics (16) @ 3.800GHz 
          `--.``               GPU: AMD ATI c2:00.0 Phoenix1 
                               Memory: 4804MiB / 27866MiB 

### Printer

Bambu A1 Mini

### How to reproduce

/usr/bin/flatpak run --branch=stable --arch=x86_64 --command=entrypoint --file-forwarding com.bambulab.BambuStudio @@u %U @@

### Actual results

[2025-09-24 17:18:15.547669] [0x00007fbebaf33480] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1

(process:2): GLib-GObject-CRITICAL **: 17:18:15.567: invalid (NULL) pointer instance

(process:2): GLib-GObject-CRITICAL **: 17:18:15.567: g_signal_handlers_disconnect_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

### Expected results

Bambu Studio runs

### Project file & Debug log uploads

[2025-09-24 17:18:15.547669] [0x00007fbebaf33480] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1

(process:2): GLib-GObject-CRITICAL **: 17:18:15.567: invalid (NULL) pointer instance

(process:2): GLib-GObject-CRITICAL **: 17:18:15.567: g_signal_handlers_disconnect_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Mujin-code, 2025-09-25:
Hello, thank you for your feedback. You can clean the configuration files of bambustudio in the config directory and restart the software. Looking forward to your reply！

cinderblockgames, 2025-09-25:
Thank you for the quick reply!  I cleared `~/.var/app/com.bambulab.BambuStudio/config` and it's working again.  I appreciate you!

closed 2025-09-25 by cinderblockgames

Mujin-code, 2025-09-25:
Okay, wish you a great experience!