bambulab/BambuStudio #8297: print speed settings window wont go away after selection

RobA80, 2025-09-26:
### Bambu Studio Version

new one

### Where is the application from?

Bambu Lab github releases, Bambu Lab Official website

### OS version

windows 11

### Printer

p1s

### How to reproduce

go to device
click print speed
once chosen
the window won't go away I have to exit the app and go back in and then its fine but it does it everytime

### Actual results

NA

### Expected results

It goes away after I choose the speed

### Project file & Debug log uploads

NA

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Mujin-code, 2025-09-28:
Hello, I'm glad for your feedback. I tried to reproduce your problem, but was unsuccessful。
To help us better understand and reproduce the problem on our side, could you please:
Record a short screen video of any steps you take. You can use a simple screen recorder such as [ScreenToGif](https://github.com/NickeManarin/ScreenToGif/releases/tag/2.41.1).
Provide the studio log: [Export Bambu PC software log](https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log)
This information will help us locate the cause more quickly and provide an effective solution. Thank you so much for your cooperation!

RobA80, 2025-09-28:
Sure no problem

Sent from my Verizon, Samsung Galaxy smartphone