bambulab/BambuStudio #8303: Impossible to enable helio additiove on 13 inch macbook

pejot-itbeton, 2025-09-27:
### Bambu Studio Version

2.2.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

macos 12.7.6

### Printer

A1 

### How to reproduce

1. Open helio
2. Try to enable
3. Cant see button

### Actual results

Modal too large, doesnt fit, cant click accept? button.

### Expected results

Should be able to see button with scrolling down or fit modal to screen.

### Project file & Debug log uploads

Not needed

### Checklist of files to include

- [ ] Log file
- [ ] Project file

qusheng-bambu, 2025-09-29:
Hello, I'm using the MAC12 system but your problem doesn't occur. Could you provide your resolution and screenshots or videos of the issue

pejot-itbeton, 2025-10-03:
Hi @qusheng-bambu ,

I have 33,3-inch (2560 × 1600) mac.