bambulab/BambuStudio #8305: Invisible Objects on Plate

nebhead, 2025-09-27:
### Bambu Studio Version

2.2.2.56 Ubuntu

### Where is the application from?

Bambu Lab github releases

### OS version

Ubuntu 24.04

### Additional system information

CPU: Intel Core i9 9900 
Memory: 32GB DDR4
GPU: nVidia GeForce 2060 (Using Nouveau Drivers) 

### Printer

Bambu P1S

### How to reproduce

Open BambuStudio
Go to Prepare
Load a model (loaded STL from local folder or from MakerLab
Model shows up as an outlined object when selected, when un-selected it is invisible

### Actual results

Model shows up as an outlined object when selected, when un-selected it is invisible

[image]

Confirmed the object is 'printable'.  

### Expected results

Expected to be able to see the object when selected or not.  

### Project file & Debug log uploads

[HollowKnight_Hornet.zip](https://github.com/user-attachments/files/22576479/HollowKnight_Hornet.zip)

Couldn't obtain the log file. 

### Checklist of files to include

- [ ] Log file
- [x] Project file

alanbernstein, 2025-10-03:
Same issue, resolved after switching to 580 proprietary, AND updating Bambu Lab from 2.0.3.54 to 2.2.1.60.

Deksor, 2025-12-14:
I'm having the exact same issue on Debian 13 with nouveau drivers and Bambu_Studio_ubuntu-24.04_PR-8834. I tried to replace the file mentioned above, but it didn't change anything.

Haidiye00, 2025-12-16:
[gouraud.zip](https://github.com/user-attachments/files/24179759/gouraud.zip)
hello,or please try this,thansk

Deksor, 2025-12-16:
> [gouraud.zip](https://github.com/user-attachments/files/24179759/gouraud.zip) hello,or please try this,thansk

Tried this, unfortunately it didn't change (I put it in the 140 folder like in your screenshot)

> Hello, could you please try installing the official NVIDIA graphics card driver instead of using Nouveau

Well I used to have official nvidia drivers, but the caused issues on my machine so I switched to nouveau and everything has been fine ... except for that minor issue

nebhead, 2026-02-12:
Just tried updating to 590 as well and also seeing good results.  I think we may actually be able to close this.

closed 2026-02-12 by nebhead