bambulab/BambuStudio #8312: AMS Pro 2: Cannot start drying when "TPU for AMS" is loaded

djtremolo, 2025-09-28:
### Bambu Studio Version

2.2.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

X1C firmware: 01.10.00.00
AMS 2 PRO firmware: 02.00.19.68

### Printer

Bambu lab X1C

### How to reproduce

1) Load ASA and "TPU for AMS" filaments into the AMS.

2) On the X1C control panel, attempt to start a drying cycle.

3) Select any drying preset (tested with PLA and others).

### Actual results

- The system refuses to start drying, showing an error that the drying temperature is not safe for TPU.
- This happens with all presets, including low-temperature ones such as PLA.
- The only way to start drying is by removing "TPU for AMS" filaments so that the AMS reports those slots as empty.

### Expected results

The drying cycle should start using the selected preset, or at minimum, presets with safe temperatures (e.g., PLA) should be allowed when "TPU for AMS" is present.

### Project file & Debug log uploads

The problem is not related to any project and it does not need bambu studio or bambu handy. The local control panel is enough. No log files either, as the problem is easy to reproduce.

### Checklist of files to include

- [x] Log file
- [x] Project file

qusheng-bambu, 2025-09-29:
Hello, currently, AMS2 pro does not support drying under feeding conditions. For more details, please refer to the wiki page

https://wiki.bambulab.com/en/ams-2-pro/ams-2-pro-for-drying-in-x1-p1-series

[image]

djtremolo, 2025-09-29:
I'm sorry, I didn't mean that. It's not about printing while drying (although that would be nice).

In my report, having a roll of "TPU for AMS" loaded in the AMS 2 Pro prevents drying completely. No printing involved.

jaekong, 2025-12-12:
This also is happening to me. Currently I am just tricking the AMS by putting TPU in after drying starts.

daneelgiskard, 2026-02-16:
I have the same issue. It won't let me dry with TPU for AMS loaded. I always have to remove the TPU from the feeder (then it won't recognize the type of filament) and only then it will let me start the standard PLA dry process at 45°.