bambulab/BambuStudio #8325: Font change not working anymore

ntreimers-beep, 2025-09-30:
### Bambu Studio Version

2.2.2.56

### Where is the application from?

Bambu Lab Official website

### OS version

OS: Windows 10 & Windows 11 (two different machines)

### Printer

BambuLab A1 (this is not involved though, the problem is in Bambu Studio)

### How to reproduce

Make text using `Bahnschrift` font and another text using `Bahnschrift Light Condensed`, notice that they look exactly the same. 

I'm not sure when this was broken, but it used to work correctly with a previous version of Bambu Studio.

### Actual results

Here is a screenshot side by side with MS Word where you can see the difference between the fonts:
[image]

### Expected results

Choosing a different font variant will show the changes properly.

### Project file & Debug log uploads

[FontIssue.zip](https://github.com/user-attachments/files/22624532/FontIssue.zip)

[logs.zip](https://github.com/user-attachments/files/22624574/logs.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

ntreimers-beep, 2025-10-02:
I confirmed that downgrading to version v02.01.01.52 of Bambu Studio does not show this problem.

cecchet, 2025-10-17:
It used to work for me on 2.2.1.60 (though the Helio Additive Optimization never showed up in the options after the upgrade from 2.0). But since the 2.3.0 update the text functionality is broken (mine defaults to Arial whatever font I put in) and it fails to modify existing text as well (it crashed randomly before but now it's just 100% broken). As a note, v 2.3.0 finally installed Helio, not sure if it's related or not.

jbast1224, 2025-10-23:
> Totally agree with jbast1224. I have had to revert to 2.01.01.52 repeatedly because the text tool is unusable for me in every subsequent version. In particular, the Bahnschrift family of fonts all display identically; there is no variation in width or weight regardless of which font is selected. I am hoping this is addressed soon so I can take advantage of other updates introduced in newer versions.

Orbitron as well. All of them appear the same on the model, and in the font selection dropdown.

LittleMountainMaps, 2025-11-19:
This issue is still present in the latest version 2.4.0.70.